MHeironimus/ArduinoJoystickLibrary #118: Ardunio pro micro load cell (hx711 lib) sim pedals wakes computer instantly.

ChrisMLian, 2019-07-04:
## Description of Issue
Sim pedals with loadcell library hx711

Leonardo is waking up computer instantly 
If i remove the axis config it will stay asleep so something to do with axis input wakes it. 
I have tested with each axis and all of them wakes the computer without touching any of the controls

## Technical Details

- Arduino Board Ardunio pro micro
- Host OS Windows 10 
- Arduino IDE Version 1.8.9

## Sketch File that Reproduces Issue

~~~
```
#include <AnalogSmooth.h>

#include <HX711.h>
#include <Joystick.h>
#define calibration_factor -2300 // Do your calibration first. default 2300
#define DOUT  3
#define CLK  2

HX711 scale;

/*

For the true false flags, use this list. Its all in  Joystick.h 
    bool includeXAxis = true,
    bool includeYAxis = true,
    bool includeZAxis = true,
    bool includeRxAxis = true,
    bool includeRyAxis = true,
    bool includeRzAxis = true,
    bool includeRudder = true,
    bool includeThrottle = true,
    bool includeAccelerator = true,
    bool includeBrake = true,
    bool includeSteering = true);
  */

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_MULTI_AXIS, 4, 3,
  false, false, true, false, false, false,
  false, true, true, true, false);

  
    
// Variable
int throttle = A0;
// int brake = 2; / no need for this as we read directly from scale
int clutch = A1;
int brake = 0;
int throttleValue = 0;
int clutchValue = 0;

// init joystick libary
void setup() {

  // Ranges are 1023 by default
   Joystick.setBrakeRange(0, 1023); 
  Joystick.setThrottleRange(0, 1023); 
  //Joystick.setZAxisRange(0, 1023);
  
  Joystick.begin();
  Serial.begin(38400);
  
  scale.begin(DOUT, CLK);
  scale.set_scale(calibration_factor); 
  scale.tare();
}

void loop() {
    //Serial.println (brake);
    //Serial.println (throttleValue);
    //Serial.println (clutchValue);
  
throttleValue = analogRead(throttle);

     if (throttleValue > 772) {
    throttleValue = 778;
  }
   else if (throttleValue < 271) {
    throttleValue = 268;
  }
    else
    throttleValue = analogRead(throttle) ;
    
  Joystick.setThrottle(throttleValue);
  //Joystick.setThrottle(analogRead(throttle));
  delay(1);

  brake = scale.get_units(); // if the value is inverted put a - sign in front like -scale.get
  
// If the value starts below 0 set it to 0. Or if its above 50 set it to 0
// This fixes the slight drift and sets it to 0 if it starts below 0
  if (brake < 0 or brake < 8) {
    brake = 0;
  }
    else
    brake = scale.get_units() ;
  Joystick.setBrake(brake);
  delay(1);

   clutchValue = analogRead(clutch);
  
  //throttle = analogRead(throttle) ; 
     if (clutchValue < 185) {
    clutchValue = 180;
  }

    else
    clutchValue = analogRead(clutch) ;
    
  Joystick.setZAxis(clutchValue);
 
 // Joystick.setZAxis(analogRead(clutch));
 
  delay(1);
}
```
~~~

## Wiring Details

Any details about pin wirings that may be relevant.

MHeironimus, 2019-07-04:
Some things you could try:

- Only call the `set` functions if the value you are setting is different from the previous value set.
- Increase the value used in the final `delay` call.

ChrisMLian, 2019-07-04:
Thank you for the quick reply

Tried different delay, no change. 

For the set functions im not 100% sure what you mean. Are you thinking of the uncommented  //Joystick.setThrottle(analogRead(throttle)); etc. Those are just remnants of the old code before i added the if else statements for HW DZ to remove any spiking. 

Its quite strange as an axis/joystick should have no bearing on the wake up calls and ive gone trough everything in terms of power and my mouse is the only thing allowed. I have also tested with only keyboard but the result is the same.

MHeironimus, 2019-07-04:
Every time you call a `set` function, a message is sent from the Arduino to the host computer with the new value. I was thinking these messages could be causing the computer to wake up (not 100% sure on this). If you persisted the last value you sent on any give `set` function and then only call `set` again if this value changes, it might prevent what you are seeing (I have not tested this theory). For example,

``` c

int lastThrottleValue = 0;

...

  if (lastThrottleValue != throttleValue) {
     Joystick.setThrottle(throttleValue);
     lastThrottleValue = throttleValue;
  }
  delay(1);
```

Disclaimer: I have not tested the code above.

ChrisMLian, 2019-07-04:
That solved it, thank you very much!  

I had to set `int lastThrottleValue = 270;` to match my number. 

`  throttleValue = analogRead(throttle);
   //throttle = analogRead(throttle) ; 
     if (throttleValue > 666) {
    throttleValue = 670;
  }
    else if (throttleValue < 273) {
    throttleValue = 270;
  }
 
    if (lastThrottleValue != throttleValue) {
     Joystick.setThrottle(throttleValue);
     lastThrottleValue = throttleValue;
  }
   delay(1);`

closed 2019-07-04 by ChrisMLian

utcoolguy, 2020-08-17:
Hi Chris, 
first of all i'm currently a complete programming noob. but i'm trying to do the same thing, I.e: create a only a load cell break pedal with Ardunio uno, hx711 and a load cell. i was wondering if i could used the Sketch File you have posted here to accomplish that. also some more help on what library will i have to include and what would be the wiring scheme required. TIA

ChrisMLian, 2020-08-17:
> Hi Chris,
> first of all i'm currently a complete programming noob. but i'm trying to do the same thing, I.e: create a only a load cell break pedal with Ardunio uno, hx711 and a load cell. i was [...]

Feel free. 

Also have a read trough this thread, It should help you get started.  : https://www.xsimulator.net/community/threads/diy-load-cell-brake-pedal-short-tuto.6042/page-4

utcoolguy, 2020-08-17:
> Feel free.
>
> Also have a read trough this thread, It should help you get started. : https://www.xsimulator.net/community/threads/diy-load-cell-brake-pedal-short-tuto.6042/page-3

Thank you for the very prompt reply, i am confused about one thing, i'm using the arduino una and i'm not sure where to plugin in the "DT" and the "SCK" from the hx711.

ChrisMLian, 2020-08-17:
> Thank you for the very prompt reply, i am confused about one thing, i'm using the arduino una and i'm not sure where to plugin in the "DT" and the "SCK" from the hx711.

Apologies i missed the UNO in your first posting. This is a Leonardo project.  You can use the loadcell on a UNO but this code and the Joystick library is for Leonardo.  I'm a beginner coder myself so i only know basic stuff. I see there are some tutorials for UNO joystick though so you could probably get it done. Afraid i wont be of much help there 

Here's and overview of the wiring anyhow. I use an Ardunio micro which is a leonardo and its the same ports.

https://www.instructables.com/id/Arduino-Scale-With-5kg-Load-Cell-and-HX711-Amplifi/

utcoolguy, 2020-08-17:
> Apologies i missed the UNO in your first posting. This is a Leonardo project. You can use the loadcell on a UNO but this code and the Joystick library is for Leonardo. I'm a beginner coder myself so [...]

yes i thing the board might be the issue. anyway thank you for you help. really appreciate it.

cagdaskan, 2020-12-30:
> yes i thing the board might be the issue. anyway thank you for you help. really appreciate it.

Hello. Did you manage to run the pedals with UNO?

Lucipy, 2021-01-05:
Hello @ChrisMLian ,
i want to build my own sim racing pedals (throttle and clutch with 10K Potis and brake with a loadcell). Unfortunately I am a complete noob in programming and I couldn't follow what you posted here. Maybe you would be so kind and help me a bit?

utcoolguy, 2021-01-05:
> Hello. Did you manage to run the pedals with UNO?

no, i actually got a leonardo for this purpose and that works fine. even the hx711 did work well for me, so made a LC amplifier to go with it.

cagdaskan, 2021-01-05:
@utcoolguy Thank you mate. I also ordered one after some failures while trying to load the program to uno. Have a nice day mate