MHeironimus/ArduinoJoystickLibrary #119: Loadcell not calibrating in Windows 10

niterida, 2019-08-03:
I am using a loadcell attached via HX711 Loadcell amp to Arduino Pro Micro for a sim racing brake.
It calibrates fine in Arduino with values over 30,000 with many kgs of pressure on the loadcell. In Windows 10 calibration it goes from 0 to 32767 at the slightest touch - like an on/off switch :(

I am not a programmer so I have just used the sketch file and wiring instructions from here : https://www.thingiverse.com/thing:3043797 

I have tried reloading the sketch and rebooting the PC with no change.

Any ideas what could be causing this or how to calibrate it properly Windows 10 ?

Thanks
Ken

Balu24, 2019-09-14:
Hello Ken,
I had a similiar issue. Try to open the Serial port before you calibrate. In my case it helped. Unfortunately you have to do that every time you want to use the Joystick.

niterida, 2019-09-15:
Thanks but I think I blew the chip – it stopped responding completely so I gave up on it altogether.

Cheers anyway 😊

Balu24, 2019-09-15:
No problem,  you could try to reset the arduino while uploading your code (only the right timing could be a little bit tricky) . 
I think it is nearly impossible to damage the bootloader. 

A detailed description how to do:

https://www.dfrobot.com/forum/viewtopic.php?t=216

niterida, 2019-09-15:
The PC is not recognising it when I plug it in so I can’t even talk to it to upload ☹

 

https://www.dfrobot.com/forum/viewtopic.php?t=216

Balu24, 2019-09-15:
If the Arduino is plugged in and you reset it (hold the reset button until the Led's blinking)  the Com Port should  appear in your Device-Manager (but only for few seconds,  don't worry about) ,  after that you should be able to select the com in Arduino IDE.  

Looked in the Code today again in my case it helped to remove the while-loop in the Void Setup function.  So i don't have to open the Serial Monitor.

niterida, 2019-09-16:
I have a Pro Micor and it doesn’t have a reset button ☹

It did flash the LEDs a couple of times when I plugged/unplugged it and I saw it flash up in the Device Manager but IDE can’t see it at all.

I think it is in the too hard basket for my limited skills.

 

Thanks anyway 😊