MHeironimus/ArduinoJoystickLibrary #131: Not visible as game controller on Windows 10

flub78, 2019-12-26:
## Description of Issue

The joystick works fine but the Leonardo is not visible as a game controller in the device manager. It is visible as a generic device and it is not possible to open the USB game controller configuration from the icon.

When I plug another joystick, I can open the USB game controller configuration panel from the icon and I see both: the other joystick and the Arduino Leonardo.

Note that it happens only on my Dell laptop, not on my desktop, both on Windows 10 64 bits. On my desktop the Leonardo is recognized as a game controller. In both cases the joystick can be used to control a game.

In spite of this detail, it is a great piece of software, making the creation of an USB joystick really easy. Thanks

## Technical Details

- Arduino Leonardo 
- Host OS Windows 10 64 bits
- Arduino IDE Version 1.8.5
- tested so far with X and Y axis, a rudder and a button

## Sketch File that Reproduces Issue

~~~
/**
   Joystick USB sur Arduino Leonardo

  Use the library Joystick by Matthew Heironimus
  https://github.com/MHeironimus
*/

#include <Joystick.h>

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_MULTI_AXIS, 32, 0,
  true, true, false, false, false, false,
  true, true, false, false, false);
  
// Constant that maps the phyical pin to the joystick button.
const int pinToButtonMap = 4;

void setup() {
  // Initialize Button Pins
  pinMode(pinToButtonMap, INPUT_PULLUP);

  Joystick.setRudderRange(400, 600);
  
  // Initialize Joystick Library
  Joystick.begin();
}

// Last state of the button
int lastButtonState = 0;

void loop() {

  // Read pin values
  int currentButtonState = !digitalRead(pinToButtonMap);
  if (currentButtonState != lastButtonState)
  {
    Joystick.setButton(0, currentButtonState);
    lastButtonState = currentButtonState;
  }

  int ailerons; //déclaration de la variable qui va stocker la valeur numérique de la tension du potentiomètre
  int profondeur;
  int palonnier;

  ailerons = analogRead(A4); 
  profondeur = 1024 - analogRead(A5); 

  // palonnier 400 - 600
  palonnier = analogRead(A3); 

  Joystick.setYAxis(profondeur);
  Joystick.setXAxis(ailerons);
  Joystick.setRudder(palonnier);
  
  delay(50);
}
~~~

## Wiring Details
Likely not relevant.

## Additional context

AlBundy33, 2020-03-08:
I have a similar issue - but in my case my arduino pro micro is recognized as game controller but does not show any buttons (I use it as button box).
But the buttons are working fine in games - so i'm not sure if this is a windows, an arduino or a library issue.

V3nom536, 2020-12-18:
Same problem as Albundy, just connected my circuit, it gets recognised by the w10 game interface but has no response from moving the axes / pressing the buttons; tested both buttons and pot for the x axis and they both work so i assume its related to a lib issue

AlBundy33, 2020-12-18:
cannot remember that i commented on this issue but my button box is working.
https://github.com/AlBundy33/ButtonBox
but havn't checked windows setting yet.

what you can try:
reset the arduino and flash it again
try other usb ports (usb 2 and 3)
uninstall drivers and reconnect

AlBundy33, 2020-12-19:
my arduino pro micros is listed as game controller.
[image]

zebolc, 2021-02-15:
I had same issue until I changed in constructor JOYSTICK_TYPE_MULTI_AXIS into JOYSTICK_TYPE_JOYSTICK,  and  I also changed first argument hidReportId into 0x06 but I'm not sure if that was necessary:
Joystick_ Joystick(0x06, 
  JOYSTICK_TYPE_JOYSTICK, 32, 0,
  true, true, false, false, false, false,
  true, true, false, false, false);

nitromancer, 2021-03-17:
I had the same issue. The problem was with the drivers windows installed. Windows recognized it as an "HID device". I had to install the "Gaming HID device" driver. (I don't know the exact English names of the drivers as my windows are in Czech.)

PatrikAlzen, 2023-01-05:
> I had same issue until I changed in constructor JOYSTICK_TYPE_MULTI_AXIS into JOYSTICK_TYPE_JOYSTICK,  and  I also changed first argument hidReportId into 0x06 but I'm not sure if that was [...]

Changing the type from JOYSTICK_TYPE_MULTI_AXIS to JOYSTICK_TYPE_JOYSTICK solved it for me.

bwanaaa, 2023-04-20:
My issue is similar but unresolved. This code compiles without error and can be loaded into an arduino leonardo:
```
// 
// Controller for dials for a  Flight Simulator 
// using an Arduino Leonardo or Arduino Micro.
// based on code from
// Matthew Heironimus
// 
//------------------------------------------------------------

#include "Joystick.h"
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_MULTI_AXIS, 32, 0,
  true, true, true, true, true, true,
  false, false, false, false, false);

char potPins[] = {
A0,A1,A2,A3,A6,A7
}; 

byte swchPins[] = {
9,8,15,14,10,16,0,1,3,2,7,5
}; 

byte potCount = 6;  
int inputPot = 0;   
byte inputSwch = 0; 
byte swchCount = 12;
int lastButtonState[12] = {0,0,0,0,0,0,0,0,0,0,0,0};

void setup() {
  Joystick.begin();
  for (byte thisSwch = 0; thisSwch < swchCount; thisSwch++) 
     pinMode(swchPins[thisSwch], INPUT);
}

void loop() {
 for (int thisPot = 0; thisPot < potCount; thisPot++) {
    inputPot = analogRead(potPins[thisPot]);
    switch (thisPot) {
     case 0:
      Joystick.setXAxis(inputPot);
      break;
     case 1:
      Joystick.setYAxis(inputPot);
      break;
     case 2:
      Joystick.setZAxis(inputPot);
      break;
     case 3:
      Joystick.setRxAxis(inputPot);
      break;
     case 4:
      Joystick.setRyAxis(inputPot);
      break;
     case 5:
      Joystick.setRzAxis(inputPot);
      break;
     default: 
      break;
    }//end switch
   }//end for

   for (byte thisSwch = 0; thisSwch < swchCount; thisSwch++)  {
  inputSwch = digitalRead(swchPins[thisSwch]);
  int currentButtonState = !digitalRead(swchPins[thisSwch]);
   /* Serial.println("currentButtonState   ");
  Serial.print(currentButtonState);
  Serial.println(inputPot);
  */
  if (currentButtonState != lastButtonState[thisSwch])
    {
      Joystick.setButton(thisSwch, currentButtonState);
      lastButtonState[thisSwch] = currentButtonState;
    }
 }//end for
}

```
Device manager shows it in the ports section:

[image](https://i.imgur.com/TPHScql.jpg)

but it is not showing up with all my other controllers in the game control panel. I think it must have something to do with the MHeironimus ArduinoJoystickLibrary. Has anyone experienced a similar issue?

zebolc, 2023-04-20:
Try to go to Control Panel -> Devices and Printers, see if you can find it there, if yes go to properties of that device, then Hardware tab, Properiees button-> Driver tab and from there try to reinstall drives, maybe it will help.

bwanaaa, 2023-04-20:
Reinstalled drives didnt work. I tried PatrikAlzen's solution and that did the trick