MHeironimus/ArduinoJoystickLibrary #132: MultipleJoystickTest exposes only 3 joystick devices

hugovangalen, 2020-01-10:
I'm trying to expose multiple joysticks but it did not work in my own code, so decided to check the supplied example.

It says on the tin it's supposed to expose 4 joysticks, but I am only seeing 3 on my machine running Ubuntu 18.04.3 LTS

Any clue as to what may be the reason for this?

MHeironimus, 2020-01-10:
Try taking a look at the Linux section of the FAQ: https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/FAQ

There is a trick to getting multiple joysticks to work in Linux.

hugovangalen, 2020-01-14:
Thanks for your response! (And I'm sorry for not actually checking the FAQ.)

I can confirm this tweak does indeed allow me to see 4 joysticks in stead of 3.

closed 2020-01-14 by hugovangalen