MHeironimus/ArduinoJoystickLibrary #133: Rename Board to custom names

Fedaykin-Wolf, 2020-02-03:
So I have 2 and 3 joysticks running on 2 separate Pro Micros. 

I was wondering if there was a way to rename the joysticks as I have 5 joysticks named "Arduino Leonardo" and it's quite hard to map the buttons in games... 

Even simple numbering would help,

Thank you for your time.

AlBundy33, 2020-03-08:
see #14

MHeironimus, 2020-06-07:
Duplicate request. See #14.

closed 2020-06-07 by MHeironimus

racdavies, 2021-02-24:
Hi, i have several button boxes(5 in total) i use for different games, but my biggest issue is with Euro truck2sim which I use 3 arduino controlled button boxes plus my Logitech g27 wheel, each is setup for specific functions.....menu navigation, cruise control options and everything else on the 3rd box, here is the issue.....in Ets2`s button config menu the game does recognize the boxes as 1, 2 and 3 but if there is a usb dropout or i switch my main Ets2 box out for my F1 2020 or Assetto Corsa Competizione  the replace the Ets2 box again and run the game all the controls are mixed up because the game cannot tell the difference, yes i can changed the order that the boxes are detected in but because the have identical names its impossible to not spend so much time having to reconfigure 44 button function, I understand Arduino want there name there but if there were the ability to add to that name permanently...... like Arduino Menu box or Arduino cruise control, or simply Arduino 1, Arduino 2 Arduino 3 then the games would be so much easier to configure and given my boxes are for my own personal use i dont see the issue.

AlBundy33, 2021-09-18:
see #14 ...