MHeironimus/ArduinoJoystickLibrary #136: Analog Inputs not working

ddragon1999, 2020-04-28:
## Description of Issue
I am a newbie when it comes to programming and I cannot get the Micro-controller to recognize the analog inputs for a 10 button, 4 axis/button D-PAD controller. All of the other positions 2-13 (except for 8, which stays on and doesn't turn off) work fine. I don't mind the position 8 not working if I can get the Analog inputs to works. I don't care how long the code gets just as long as it work. Can somebody please help me out with what to add/remove to get the Analog buttons to work. I would greatly appreciate it

## Technical Details

- RedBear Blend V1 (ATmega32U4, Nordic nRF8001)
- Windows 10
- Arduino IDE Version 1.8.12

## Sketch File that Reproduces Issue

~~~
#include "Joystick.h"

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_GAMEPAD,
  11, 0,                  // Button Count, Hat Switch Count
  true, true, false,     // X and Y, but no Z Axis
  false, false, false,   // No Rx, Ry, or Rz
  false, false,          // No rudder or throttle
  false, false, false);  // No accelerator, brake, or steering
  
void setup() {
  // Initialize Button Pins
  pinMode(2, INPUT_PULLUP);
  pinMode(3, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(5, INPUT_PULLUP);
  pinMode(6, INPUT_PULLUP);
  pinMode(7, INPUT_PULLUP);
  pinMode(8, INPUT_PULLUP);
  pinMode(9, INPUT_PULLUP);
  pinMode(10, INPUT_PULLUP);
  pinMode(11, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
  pinMode(13, INPUT_PULLUP);
  pinMode(A0, INPUT);
  pinMode(A1, INPUT);

  
  // Initialize Joystick Library
  Joystick.begin();
  Joystick.setXAxisRange(-1, 1);
  Joystick.setYAxisRange(-1, 1);
}

// Last state of the buttons
int lastButtonState[14] = {0,0,0,0,0,0,0,0,0,0,0,0,0,0};

void loop() {

  // Read pin values
  for (int index = 0; index < 14; index++)
  {
    int currentButtonState = !digitalRead(index + 2);
    if (currentButtonState != lastButtonState[index])
    {
      switch (index) {
        case 0: // UP
          if (currentButtonState == 1) {
            Joystick.setYAxis(-1);
          } else {
            Joystick.setYAxis(0);
          }
          break;
        case 1: // RIGHT
          if (currentButtonState == 1) {
            Joystick.setXAxis(1);
          } else {
            Joystick.setXAxis(0);
          }
          break;
        case 2: // DOWN
          if (currentButtonState == 1) {
            Joystick.setYAxis(1);
          } else {
            Joystick.setYAxis(0);
          }
          break;
        case 3: // LEFT
          if (currentButtonState == 1) {
            Joystick.setXAxis(-1);
          } else {
            Joystick.setXAxis(0);
          }
          
          break;
          case 4: // TRIANGLE
          Joystick.setButton(0, currentButtonState);
          break;
        
          break;
          case 5: // SQUARE
          Joystick.setButton(1, currentButtonState);
          break;

          break;
          case 6: // CROSS
          Joystick.setButton(2, currentButtonState);
          break;

          break;
          case 7: // CIRCLE
          Joystick.setButton(3, currentButtonState);
          break;

          break;
          case 8: // L1
          Joystick.setButton(4, currentButtonState);
          break;

          break;
          case 9: // L2
          Joystick.setButton(5, currentButtonState);
          break;

          break;
          case 10: // R1
          Joystick.setButton(6, currentButtonState);
          break;

          break;
          case 11: // R2
          Joystick.setButton(7, currentButtonState);
          break;
          
          break;
          case 12: // Select
          Joystick.setButton(8, currentButtonState);
          break;
          
          break;
          case 13: // Start
          Joystick.setButton(9, currentButtonState);
          break;

      }
       lastButtonState[index] = currentButtonState;
    }
  }
}
~~~

## Wiring Details
Pin Signal Wiring are as follows:
// Pin 2 = UP
// Pin 3 = RIGHT
// Pin 4 = DOWN
// Pin 5 = LEFT
// Pin 6 = TRIANGLE
// Pin 7 = SQUARE
// Pin 8 = CROSS
// Pin 9 = CIRCLE
// Pin 10 = L1
// Pin 11 = L2
// Pin 12 = R1
// Pin 13 = R2
// Pin A0 = START
// Pin A1 = SELECT
With the connector ground going to micro-controller ground

## Additional context

kingfisher09, 2020-05-09:
My code for an analogue stick which seems to work is very simple. I don't actually define my pins in (this may be bad practice I'm new to Arduino) but it seems to work. The calculation I'm doing to get x and y is to convert from the 0-1024 that the Arduino reads to the -127 - +127 that Joystick asks for. 

Hope this helps. 

```
void loop() {
  // put your main code here, to run repeatedly:
  byte x = round(((analogRead(A3)) - 512) * (127.0/512));
  byte y = round(((analogRead(A2)) - 512) * (127.0/512));
  Joystick.setXAxis(x);
  Joystick.setYAxis(y);
}

targodan, 2020-05-09:
Been a while since I played with this library but I have some notes.

- You should try calibrating the analog input via the windows calibration.
- I'm not sure what kind of data range windows expects but it's certainly more than -1 to 1. Might be calibration is sufficient, otherwise try a different range like -127 to +127 and setting that when the buttons are pressed.
- Output the values you give to the setXAxis/setYAxis function via Serial to verify that it's actually a problem with this lib and not just a wiring problem.

Just a hint @kingfisher09, try to avoid floatingpoint operations when possible. They are really slow on arduino as it doesn't have hardware support for floats. Reordering of the operations let's you do it entirely in int. `x = (analogRead(A3) - 512) * 127 / 512` should do the job. (again, output via Serial to verify)

kingfisher09, 2020-05-09:
@targodan I tried serial outputting the result of 127/512 and it gave 4 rather than 4.03. Most likely that level of accuracy would be fine tbh but it is different.

targodan, 2020-05-09:
True, it's a bit different. But the lib uses int anyway.
So with the float method and rounding your error is +/- 0.5, and in pure int way the error -0 / +1.

Same range of error, slightly offset. You'll have quantization error coming from the adc anyway, so you don't really gain much. Then again in this application the speed difference really doesn't matter. :man_shrugging: 

Just thought I'd mention it cause I fell in that trap once, where I the code had to be fairly efficient to be quick enough and getting rid of any float stuff sped it up a hole bunch (plus the binary size is also lower without floats).

kingfisher09, 2020-05-17:
@targodan, thanks for the tips!

Since you've been so helpful, I wonder if you could help with a separate problem I'm having? 

The problem is basically that I don't understand properly how to use the constructor to change the joystick settings. I've made a post here that explains my problems;
https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/140

Thanks!

closed 2020-06-25 by MHeironimus

CryptoAngel, 2020-07-06:
Hi,
Can send the full code that worked for you?
I need Joystick code and wiring for 8 buttons, 3 potentiometers, HAT (a button with 4 directions), and if possible to add FFB (saw someone submit sample here).
Can you help?