MHeironimus/ArduinoJoystickLibrary #137: Gamepad connection to games

fooff, 2020-04-30:
## Description of Issue
Hello, i used your library to create a gamepad that correctly works as windows correctly detect it and i can see the joystick and button input reach it.
However, i tried with many different game, and no one is able to recognize it yet.
Is there any particular process i have to follow in order for games to recognize a custom gamepad?

Could the fact that i'm using the Keyboard library important for this issue?

## Technical Details

- Arduino Board: Micro 
- Host OS: Windows 10
- Arduino IDE Version: 1.8.12

## Sketch File that Reproduces Issue
#include <Joystick.h>
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_GAMEPAD, 32, 0, true, true, false, false, false, false, false, false, false, false, false);
~~~
Sketch file goes here (if applicable)
[test_joystick.txt](https://github.com/MHeironimus/ArduinoJoystickLibrary/files/4557881/test_joystick.txt)

~~~

## Wiring Details

Not relevant

kingfisher09, 2020-05-09:
I have the same issue, please post here if you solved the issue.

fooff, 2020-05-21:
sadly not yet, for now i surrendered to use a WASD configuration instead

TheUNkilled123, 2020-05-28:
You just need to use an Xinput wrapper like x360ce for your games. Look up how to use it online and it should work like a charm!

VkTheProgrammer17, 2021-01-23:
it should work with steam