MHeironimus/ArduinoJoystickLibrary #140: Error compiling for board Arduino Micro.

kingfisher09, 2020-05-09:
## Description of Issue
This is probably a really easy fix but I'm new to Arduino and it's got me stumped. 

I'm having problems creating the Joystick constructor using Joystick_ Joystick. I have tired a few different things as listed below.

At first my constructor looked like this:
```
//Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_GAMEPAD, 
10, 2, true, true, false, false, false, false, false, false, false, false, false);
```

However that gave the error: 'JOYSTICK_DEFAULT_REPORT_ID' was not declared in this scope

So I tried:
```
Joystick_ Joystick(0x03, 0x05, 
10, 2, true, true, false, false, false, false, false, false, false, false, false);
```

Which gave the error: no matching function for call to 'Joystick_::Joystick_(int, int,......

So then I tried just:
```
Joystick_ Joystick;
```
Which gave the error: Error compiling for board Arduino Micro.

Finally I tried 
```
Joystick_ Joystick();
```

Which gave the error: 'Joystick_ Joystick()' redeclared as different kind of symbol

So now I'm really puzzled. What am I doing wrong?

## Technical Details

- Arduino Pro Micro
-  Windows 10:
- Arduino IDE Version 1.8.12

~~~
Full sketch showing a few attempts:

```

#include <Joystick.h>
//Joystick_ Joystick(0x03, 0x05, 10, 2, true, true, false, false, false, false, false, false, false, false, false);

Joystick_ Joystick();

int Rb = 16;
int Rt = 10;

String randname;
  
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Joystick.begin();
  pinMode(Rb, INPUT_PULLUP);
  pinMode(Rt, INPUT_PULLUP);
}

void loop() {
  // put your main code here, to run repeatedly:
  byte x = round(((analogRead(A3)) - 512) * (127.0/512));
  byte y = round(((analogRead(A2)) - 512) * (127.0/512));
  Joystick.setXAxis(x);
  Joystick.setYAxis(y);

  if (digitalRead(Rb) == LOW) {
    Joystick.setZAxis(0);
  } else {
    Joystick.setZAxis(180);
  }
  
  if (digitalRead(Rt) == LOW) {
    Joystick.pressButton(5);
  } else {
    Joystick.releaseButton(5);
  }
}

~~~

targodan, 2020-05-17:
Not sure why it's not working. Could be a problem related to the Arduino Micro board?

This will never work: `Joystick_ Joystick();` because C++ interprets this as a function definition (or a function pointer, I'm not 100% sure). But the other variants should all work. What're the error messages you get for the `Joystick_ Joystick;` variant?

kingfisher09, 2020-05-17:
I get:

``Error compiling for board Arduino Micro.``

I seem to get this error whatever board I try to compile for. I also tried compiling for the Leonardo and the Uno...

kingfisher09, 2020-05-17:
I found that if I move the constructor to the setup then it will compile. However when I did that, the Arduino crashed every time it ran.

kingfisher09, 2020-05-17:
Ok, I've found the problem. I'm an idiot and I seem to have downloaded an old version of the repository. Thanks for you help!

closed 2020-06-07 by MHeironimus