MHeironimus/ArduinoJoystickLibrary #141: Joystick.end() does nothing

Bustean, 2020-05-09:
## Description of Issue

I experimented with the use of the Joystick.end()  function, but it does not stop windows from seeing the device, and as such some games will automatically use the board as an input (in particular trackmania turbo, which has steering and throttle on split axes) meaning that when i start the game the car automatically accelerates and steers left (because i also reset the axes to 0 when i call the function), despite having used the function Joystick.end().

## Technical Details

- Arduino micro (32u4)
- Windows 10
- Arduino IDE Version 1.8.12:

## Sketch File that Reproduces Issue

```
#include "Joystick.h"
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,
                   JOYSTICK_TYPE_MULTI_AXIS, 0, 0,
                   false, false, false, false, false, false,
                   false, true, true, true, false);
int gas = 0;
int brake = 0;
int clutch = 0;

void setup() {

  Serial.begin(115200);
  Joystick.setThrottleRange(0, 260);
  Joystick.setAcceleratorRange(0, 260);
  Joystick.setBrakeRange(0, 260);

  Joystick.begin(false);

  Joystick.setThrottle(0);
  Joystick.setAccelerator(0);
  Joystick.setBrake(0);

  pinMode(A2, INPUT);
  pinMode(A3, INPUT);
  pinMode(A1, INPUT);
  pinMode(2, INPUT_PULLUP);
  pinMode(3, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(5, INPUT_PULLUP);
}

void loop() {
  bool once = true;
  while (digitalRead(2) == false) {
    if (once == true) {
      Joystick.setThrottle(0);
      Joystick.setAccelerator(0);
      Joystick.setBrake(0);
      once = false;
    }
    Joystick.end();
  }
  Joystick.begin();

  gas = map(analogRead(A3), 227, 340, 0, 255);
  Joystick.setThrottle(gas);

  clutch = map(analogRead(A2), 629, 715, 0, 255);
  Joystick.setAccelerator(clutch);

  brake = map(analogRead(A1), 237, 800, 0, 255);
  Joystick.setBrake(brake);

  if (!digitalRead(5) == true) {
    Serial.print("Clutch ");
    Serial.print(analogRead(A2));
    Serial.print("   ");
    Serial.println(clutch);
  }

  if (!digitalRead(4) == true) {
    Serial.print("Brake ");
    Serial.print(analogRead(A1));
    Serial.print("   ");
    Serial.println(brake);
  }
  if (!digitalRead(3) == true) {
    Serial.print("Gas ");
    Serial.print(analogRead(A3));
    Serial.print("   ");
    Serial.println(gas);
  }
  Joystick.sendState();
}
```

## Wiring Details

Pins 2, 3, 4 and 5 are connected to 4 dip switches, to control the serial transmission for each pedal, while the 4th one puts the code in an endless while loop so that the board won't wake up the pc during sleep.

The analog pins A0, A1 and A2 connect to the pedals themselves.
## Additional context

I tried moving the .begin, .set...Range functions around and making a function that resets the board, such that whenever i flip the switch connected to pin 2, everything would stop. I had no success with this either, as it seemed almost as if windows started recognizing the device from the declaration of the object itself.

alijani1, 2020-05-17:
This would be helpful as currently there is no way to remove dynamically a controller that was detected to be disconnected.  One use case I was trying was when i detect paddle controllers plugged in instead of joystick, I wanted to remove the existing joystick that was previously connected and replace with paddles.

VkTheProgrammer17, 2021-02-07:
you might want to re-install the Joystick library