MHeironimus/ArduinoJoystickLibrary #144: The Joystick library can only be used with a USB MCU

MitchK, 2020-05-24:
## Description of Issue

Hi, I'm very new to Arduino development. When uploading the code to my board, I get this error:

> The Joystick library can only be used with a USB MCU (e.g. Arduino Leonardo, Arduino Micro, etc.).

It seems like the library does only support Arduino Leonardo or Micro. Is there a way/workaround to get it to run on a Mega 2560 or is there no way around buying a new board?

Thank you!

## Technical Details

- Arduino Board (e.g. Arduino Leonardo): Mega 2560
- Host OS (e.g. Windows 10): Windows 10
- Arduino IDE Version (e.g. 1.8.3):

## Sketch File that Reproduces Issue

JoystickTest sketch

## Wiring Details

N/A

MHeironimus, 2020-05-24:
That is correct. This library is only supported on the cards listed in the wiki: https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/Supported-Boards

MitchK, 2020-05-25:
Thanks! Just got my Leonardo today, closing this.

closed 2020-05-25 by MitchK