MHeironimus/ArduinoJoystickLibrary #145: Can't add the library

qwertyalfa, 2020-05-24:
When I try to add the library from ZIP / folder, I cant do so. I get information that there is no correct library in that directory. Can anyone help me?

CryptoAngel, 2020-06-03:
To which folder you're trying to copy the file? Have extract it? The folder is read only? What error message you see?

qwertyalfa, 2020-06-03:
I managet to fix it, but thanks for a reply :)
Something was odd with some of my drivers, after updating it works like a charm.

CryptoAngel, 2020-06-04:
By the way, do you know if this is the most updated version of ArduinoJoystick?
I building my own joystick and I need X,Y,Z and those buttons and the hat buttons but I am not sure this is the perfect solution. Also there he did not include the wire connections...
Regards

closed 2020-06-07 by MHeironimus