MHeironimus/ArduinoJoystickLibrary #147: Is this library still maintained?

SukkoPera, 2020-06-07:
- No updates in 3 years
- 60 pending issues
- 10 pending pull requests

All of the above make me think that this library is no longer maintained, which is a real pity, since it works very well and it seems to be used by **a lot** of people.

If this is the case, please at least state it officially, so that hopefully a new maintainer can be found, or people can switch to a different library.

MHeironimus, 2020-06-07:
Yes, it is still being maintained. I would love to work on it more than I do, but I have other responsibilities and priorities.

I plan on approving a lot of the pull request, I just need to find time to test them before I make them official. Anyone is welcome to use the Pull Request versions, they just need to realize they have not been fully tested yet.

A lot of the pending issues need to be closed. There is a pretty active community that answers questions that come up. I am thankful for their help as I do not always have time to answer every question that comes in.

closed 2020-06-07 by MHeironimus

SukkoPera, 2020-06-07:
This is great to hear. Keep it up then! :)