MHeironimus/ArduinoJoystickLibrary #148: Mouse moves when axis move on joystick

eddo888, 2020-06-09:
## Description of Issue
I have a simple joystick configuration, I map an analog input to a joystick input.
I am not including anything other than the joystick.h
when I move the joystick axis, it moves the mouse

## Technical Details
- Arduino Board = Arduino Leonardo
- Host OS = MacOS Mojave, 10.14.6
- Arduino IDE Version = 1.8.12

## Sketch File that Reproduces Issue

~~~
// Program used to test the USB Joystick library when used as 
// a Flight Controller on the Arduino Leonardo or Arduino 
// Micro.
//
// Matthew Heironimus
// 2016-05-29 - Original Version
//------------------------------------------------------------

#include "Joystick.h"

Joystick_ Joystick(
  JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_MULTI_AXIS,
  32,
  0,
  true,  // Xaxis
  true,  // Yaxis
  true,  // Zaxis
  false, // rXaxis
  false, // rYaxis
  false, // rZaxis
  true,  // Rudder
  true,  // Throttle
  true,  // Accellerator
  false, // Brake
  false  // Steering
);

static const uint8_t D2 = 2;
static const uint8_t D3 = 3;
static const uint8_t D4 = 4;
static const uint8_t D5 = 5;
static const uint8_t D6 = 6;

// Mapping of analog pins as digital I/O
/*
static const uint8_t A0 = 18;
static const uint8_t A1 = 19;
static const uint8_t A2 = 20;
static const uint8_t A3 = 21;
static const uint8_t A4 = 22;
static const uint8_t A5 = 23;
*/

// A6-A11 share with digital pins
/*
static const uint8_t A6 = 24;  // D4
static const uint8_t A7 = 25;  // D6
static const uint8_t A8 = 26;  // D8
static const uint8_t A9 = 27;  // D9
static const uint8_t A10 = 28; // D10
static const uint8_t A11 = 29; // D12
*/

void setup() {
  
  pinMode(D2, INPUT); // bit4
  pinMode(D3, INPUT); // bit3
  pinMode(D4, INPUT); // bit2
  pinMode(D5, INPUT); // bit1

  pinMode(D6, INPUT); //button
  
  Joystick.setXAxisRange(-127, 127);
  Joystick.setYAxisRange(-127, 127);
  Joystick.setZAxisRange(-127, 127);
  
  Joystick.setRudderRange(-127, 127);
  Joystick.setThrottleRange(0, 255);
  
  Joystick.setAcceleratorRange(0, 255);
  //Joystick.setBrakeRange(0, 255);
  //Joystick.setSteeringRange(-127, 127);
  
  Joystick.begin(false);

}

int xAxis = -1;
int yAxis = -1;
int zAxis = -1;
int rudder = -1;
int throttle = -1;
int accelerator = -1;

int bit1 = -1;
int bit2 = -1;
int bit3 = -1;
int bit4 = -1;
uint8_t counter = -1;

uint8_t button = -1;

int logging = 0;

void loop() {

  delay(1); // delay  between reads for stability
  
  char mystr[16];

  bit4 = digitalRead(D2);
  bit3 = digitalRead(D3);
  bit2 = digitalRead(D4);
  bit1 = digitalRead(D5);

  uint8_t counterI = 8*bit4 + 4*bit3 + 2*bit2 + 1*bit1;

  if (counterI != counter) {
    if (logging) {
      sprintf(mystr, "Counter = %d %d%d%d%d", counterI, bit4, bit3, bit2, bit1);
      Serial.println(mystr);
    }
    
    if (counter >= 0) {
      Joystick.releaseButton(counter);
    }  
    
    counter = counterI;
    
    if (counter < 32) {
      Joystick.pressButton(counter);
    }

  }

  uint8_t buttonI = digitalRead(D6);
  if (logging && buttonI != button) {
    sprintf(mystr, "Button = %d", buttonI);
    Serial.println(mystr);
    button = buttonI;
  }
  Joystick.setButton(32-1, buttonI);
  
  int xAxisI = analogRead(A0);
  float xAxisF = (xAxisI*256/1024)-127;
  if (logging && xAxisI != xAxis) {
    sprintf(mystr, "Axis X=%d", xAxisI);
    Serial.println(mystr);
    xAxis = xAxisI;
  }
  Joystick.setXAxis(xAxisF);

  int yAxisI = analogRead(A1);
  float yAxisF = (yAxisI*256/1024)-127;
  if (logging && yAxisI != yAxis) {
    sprintf(mystr, "Axis Y=%d",yAxisI);
    Serial.println(mystr);
    yAxis = yAxisI;
  }
  Joystick.setYAxis(yAxisF);

  int zAxisI = analogRead(A2);
  float zAxisF = (zAxisI*256/1024)-127;
  if (logging && zAxisI != zAxis) {
    sprintf(mystr, "Axis Z=%d",zAxisI);
    Serial.println(mystr);
    zAxis = zAxisI;
  }
  Joystick.setZAxis(zAxisF);

  int rudderI = analogRead(A3);
  float rudderF = (rudderI*256/1024)-127;   
  if (logging && rudderI != rudder) {
    sprintf(mystr, "Axis rudder=%d",rudderI);
    Serial.println(mystr);
    rudder = rudderI;
  }
  Joystick.setRudder(rudderF);

  int throttleI = analogRead(A4);
  float throttleF = (throttleI*256/1024);
  if (logging && throttleI != throttle) {
    sprintf(mystr, "Axis throttle=%d",throttleI);
    Serial.println(mystr);
    throttle = throttleI;
  }
  Joystick.setThrottle(throttleF);

  int acceleratorI = analogRead(A5);
  float acceleratorF = (acceleratorI*256/1024);
  if (logging && acceleratorI != accelerator) {
    sprintf(mystr, "Axis accelerator=%d",acceleratorI);
    Serial.println(mystr);
    accelerator = acceleratorI;
  }
  Joystick.setAccelerator(acceleratorF);

/*
  int brakeI = analogRead(A6); // D4
  float brakeF = (brakeI*256/1024);
  sprintf(mystr, "Axis brake=%d",brakeI);
  //Serial.println(mystr);
  Joystick.setBrake(brakeF);

  int steeringI = analogRead(A7); // D6
  float steeringF = (steeringI*256/1024)-127;
  sprintf(mystr, "Axis steering=%d",steeringI);
  //Serial.println(mystr);
  Joystick.setSteering(steeringF);
*/

  Joystick.sendState();

}
~~~

## Wiring Details
Any pin wiring details that may be relevant.

## Additional context
Add any other context about the problem here.

eddo888, 2020-06-09:
If I change the joystick type to "JOYSTICK_TYPE_JOYSTICK" the issue goes away.
```
Joystick_ Joystick(
  JOYSTICK_DEFAULT_REPORT_ID, 
  //JOYSTICK_TYPE_MULTI_AXIS,
  JOYSTICK_TYPE_JOYSTICK,
  32,
  0,
  true,  // Xaxis
  true,  // Yaxis
  true,  // Zaxis
  false, // rXaxis
  false, // rYaxis
  false, // rZaxis
  true,  // Rudder
  true,  // Throttle
  true,  // Accellerator
  false, // Brake
  false  // Steering
);
```

MHeironimus, 2020-06-09:
Looks like you found a solution to your issue, so I will close this. I have observed this behavior on Linux machines when the `joystickType` is set to `JOYSTICK_TYPE_MULTI_AXIS` as well.

closed 2020-06-09 by MHeironimus