MHeironimus/ArduinoJoystickLibrary #149: Add support for Android/iOS devices

tomxposed, 2020-06-24:
The library **does not work on my android phone**. I don't know whether it is a bug or not. If it is not implemented please do so, it will be very useful for games like PUBG and COD

BusteanHAN, 2020-06-24:
The library works on the USB HID standard which is not supported by phones. The whole point of HID is to not use drivers, but to connect an input device to any phone you need a driver.

tomxposed, 2020-06-25:
The keyboard library used is perfectly working on the android devices

closed 2020-06-25 by tomxposed

tomxposed, 2020-07-04:
@BusteanHAN I have implemented the library for my [iBus-Joystick](https://github.com/tomxposed/iBus-Joystick#ibus-joystick) project and it works fine in Android, iOS, Mac, and Windows machines without any drivers :-)