MHeironimus/ArduinoJoystickLibrary #153: Encoders to build radios and cockpits

ESA-Angel, 2020-07-04:
Would it be possible to add support for encoders?
They are very useful to build radios for airplane simulation, and right now the only advantage of using Leo Bodnar
This would allow building cockpits for a fraction of the cost of a LB

faragher, 2020-07-29:
Encoders like rotary encoders where you have a dial with multiple values? How does the software interpret the input? If each input is a button, then just wire up each position to an input. If it's continuous, hook it up to an axis.

Which program are you using? DCS?

Reisen-Wandern-Tauchen, 2020-07-31:
I guess ESA-Angel was asking for support of rotary encoders. Visually looking like rotary potentiometers and basically consisting of 2 switches that are phase shifted by 90°. Therefore, the two switches can take the following status combinations: 0/0, 1/0, 1/1, 0/1; rotation direction can be detected by the change of previous and current status of the switches.

faragher, 2020-07-31:
Isn't that hardware implementation specific? I'd find libraries interesting, but isn't this a framework, not an implementation? I'm not against making a hardware/firmware solution for this, but I think it's out of scope for the library.

ESA-Angel, 2020-08-02:
Hello again, in fact I was talking about rotary encoders, they rotate 360º without end (some also have a button)

I use DCS and with DCS Bios I have support but it only works for that simulator and a single plane.

There is also a firmware called MMJoy2 for Arduino boards that transforms them into a HiD but it is not compatible with Arduino Due and to manage pins you need shift registers.

This feature would allow the construction of generic radios for airplanes of all kinds of simulators.

faragher, 2020-08-02:
I get what you're looking for now. 

It would be interesting to dig into how MMJoy sends encoder data and how the program interprets it, but I think you'd be disappointed anyway. You need shift registers to get a significant number of buttons using this software as well, and to the best of my knowledge there is no GUI to set it up, you have to directly program the loops and event handling into the microprocessor. 

If I had the hardware, I'd give it a shot, but you'd still need to program the handler and build the hardware interface, which would include custom circuit boards and shift registers unless you wanted to run only a few buttons per microcontroller.

Reisen-Wandern-Tauchen, 2020-08-03:
@ESA-Angel: Maybe this article would be of some use how to implement a "decoding routine" for rotary encoders: https://www.electronicsweekly.com/blogs/engineer-in-wonderland/rotary-encoder-decode-arduino-2018-01/
Keep to the example with just one encoder. It's basically working with a two-dimensional matrix array (4x4) that defines the possible "neighbouring" steps to any of the 4 possible states. Keep in mind that this array is working on a 2-bit binary system not decimal!

markus-i, 2020-10-25:
In case you're still looking: Paul Stoffregen's lib may be helpful: https://github.com/PaulStoffregen/Encoder

ESA-Angel, 2020-10-26:
Thanks everyone, in the end I used mmjoy2 for my project (a Huey collective for simulators) but I'm saving this for another time

https://github.com/MMjoy/mmjoy_en/wiki

closed 2021-02-28 by MHeironimus

ttait-vantim, 2021-03-15:
If you want to see an example of rotary encoders via the https://github.com/PaulStoffregen/Encoder library and this ArduinoJoystickLibrary being used you can check out my throttle quadrant on ThingiVerse. The code is attached to the thing.

https://www.thingiverse.com/thing:4699317

Tim