MHeironimus/ArduinoJoystickLibrary #154: Rudder and brake do not work at the same time

brunombc, 2020-07-05:
## Description of Issue
I'm making a rudder pedal with brake, but when I set Rudder and Brake in the code, only Brake is available in the game controller properties and it work, but not the rudder. 

## Technical Details

- Arduino Board (e.g. Arduino Leonardo): 
- Host OS (e.g. Windows 10):
- Arduino IDE Version (e.g. 1.8.3):

## Sketch File that Reproduces Issue

```
#include "Joystick.h"

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_MULTI_AXIS, 0, 0,
  false, false, false, false, false, false,
  true, false, false, true, false);

int rudPed = A0;
int rudValue = 0;

int brakePed = A1;
int brakeValue = 0;

void setup() {
  Joystick.begin();
}

void loop () {

  rudValue = analogRead(rudPed);
  Joystick.setRudder(rudValue);
  delay(1);

  brakeValue = analogRead(brakePed);
  Joystick.setBrake(brakeValue);
  delay(1);
}
```

## Additional context

When I set Throttle instead of Brake, it works normally.
How can I set Rudder and Brake instead of Rudder and Throttle?

FlisherOfatale, 2020-07-05:
Similar issue with Z cauing the Joystick to not be detected by windows if RX or RY are enabled.
Same issue if we try to add RZ and X + Y

rafaelmaiolla, 2020-08-15:
I had a similar issue on Windows 10.
I was trying to have Rudder, Throttle, Accelerator and Brake. But Rudder didn't work.

Testing the same code on Linux seems to work fine.  I also tested using the Gamepad api from the browser (Chrome) and it is working as expected on Linux and Windows - https://gamepad-tester.com/

I'm not sure if Windows has any limitation regarding the combination you use in your joystick.

For now, I'm using Steering instead of Rudder and Windows is accepting it. However Steering + Accelerator become a kind of a x,y axis.

ALEEF02, 2020-09-16:
Having the same issue...

batbsv, 2020-12-18:
https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/177

VkTheProgrammer17, 2021-01-23:
Just use another one of the axes which is not already taken

VkTheProgrammer17, 2021-01-23:
write down which axis does what function and you'll be okay

VkTheProgrammer17, 2021-02-07:
or you could re-calibrate everything

Thick8, 2023-01-29:
I tried your code and the rudder is not showing up in the Windows test. Everything else does though. I'm wondering if there is an issue with setRudder. 

```
#include "Joystick.h"
                                                      
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_JOYSTICK, 
0, 0, false, false, false, true, true, true, false, false, false, false, false);                 

//int rudPed = A0;
//int rudValue = 0;

int Rx = A0;
int RxValue = 0;

int Ry = A1;
int RyValue = 0;

int Rz = A2;
int RzValue = 0;

void setup() {

  Joystick.setRxAxisRange(0, 1023);
  //Joystick.setRudderRange(-512, 512);
  Joystick.setRyAxisRange(0, 1023);
  Joystick.setRzAxisRange(0, 1023);
  
  Joystick.begin();
}

void loop() {

  RxValue = analogRead(Rx);
  Joystick.setRxAxis(RxValue);
  delay(1);

  //rudValue = analogRead(rudPed);
  //Joystick.setRudder(rudValue);
  //delay(1);

  RyValue = analogRead(Ry);
  Joystick.setRyAxis(RyValue);
  delay(1);

  RzValue = analogRead(Rz);
  Joystick.setRzAxis(RzValue);
  delay(1);
}
 
```

Amy-81, 2024-12-28:
Hi, I had same issue. I'm building rudder pedals with differential brakes (for flight sims). I've done some testing today and found that in Win 10 each Joystick library axis is recognized as follows (regardless of their names on Game Controller screen):

X, Y, Z - axis X, Y, Z
Rx, Ry, Rz - rotation Rx, Ry, Rz
rudder - rotation Rz
throttle - slider S0
accelerator - axis Y
brake - rotation Rz
steering - axis X

You can clearly see that Rz, Rudder and Brake are assigned (at least in Windows 10) to same axis as rotation Rz. That's why you can't use them at same time.

I already tested setup with 3 potentiometers set as Accelerator (Left brake), Brake (Right brake) and Steering (Rudder). Setup as X, Y, Rz or others work as well, as long as you do not use same axis more then once. In my game this setup was recognized as Joy1 : axis 1, Joy1 : axis 2 and Joy 1 : axis 3.

It would be nice to have L_brake (slider S0) and R_brake (slider S1) to use in the library. This way it would be easy to build one flight sim controller having joystick (axis X, Y), throttle (axis Z), rudder (rotation Rz), left brake (S0), right brake (S1) and we still have Rx and Ry to use as fuel mixture and propeller pitch controls if needed. That will use 8 axes that Windows can handle.

Amy