MHeironimus/ArduinoJoystickLibrary #155: Legs map / mapping

CryptoAngel, 2020-07-20:
Hi,

Why A0 is ground, isn't "ground" suppose to be?
Can you see the picture attached, why isn't it working?

Regards
[image]

MHeironimus, 2020-07-22:
I am sorry. I do not understand the question.

CryptoAngel, 2020-07-22:
How to connect the analog potentiometer for flight joystick?

MHeironimus, 2020-07-22:
You will need to connect the X axis and Y axis outputs of your stick to analog inputs on the Arduino. You will then need to read those values in using the `analogRead` function and sent the results to the PC via the Arduino Joystick library's `setXAxis` or `setYAxis` functions. Which pins you actually use is based on how you write your sketch file.

CryptoAngel, 2020-07-22:
I'm using your example with your library. you stated that ground suppose to be A0 but you can see on the picture I uploaded there's dedicated "ground" on the Leonardo board. The wiring on the picture is not correct?

MHeironimus, 2020-07-22:
I think you have misunderstood the purpose of the `FlightControllerTest` example. It is not an example of how to make your own Flight Controller (with real hardware). It is a test script verifying the functionality of the Joystick library. Grounding `A0` causes the test script to execute (or pause if you let the A0 pin float high). You will need to write your own sketch file to accomplish what you are trying to do.

CryptoAngel, 2020-07-22:
Do you have an example code for Joystick that use your library with analog XY and digital button and maybe hat?

MHeironimus, 2020-07-26:
A list of example projects people had done with the library has recently been added to the wiki ([https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/Examples](https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/Examples)).

closed 2020-07-26 by MHeironimus