MHeironimus/ArduinoJoystickLibrary #158: Device type is always "Driving"

MaciejNiklasinski, 2020-08-19:
## Description of Issue
Not sure whether that's a bug report or a feature request. So regardless of what constant I will use when initializing the Joystick (JOYSTICK_TYPE_JOYSTICK/JOYSTICK_TYPE_GAMEPAD/JOYSTICK_TYPE_MULTI_AXIS) the created the device will always be of type "Driving". Almost all programs are still able to use it as normal joystick but FSUIPC beta 7, an addon to new Microsoft Flight Simulator which I intended to use to help me to map all the controls, unfortunately, doesn't (It does detect all my 'real' joysticks).

[image]

Is that a bug or the JOYSTICK_TYPE_* constants specifying something else and the intended HID type is "Driving"? If that's not a bug I would like to request a feature, please. I tried to look into the DynamicHID code, I know C# and JavaScript but my understanding of c++ is very slim (as you will see by looking into the attached sketch, I do realize that I follow possibly anti-pattern there but I just wanted something to work quickly and couldn't spend time on learning the language properly) so I can't see what would need to be changed if anything to change the HID type there.

## Technical Details

- Arduino Leonardo 
- Windows 10
- 1.8.13

## Sketch File that Reproduces Issue
```

#include <SPI.h>
#include <Joystick.h>

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_MULTI_AXIS, 128, 0);

const byte LATCH = 0;

const uint8_t NUMBER_OF_SHIFT_REGISTERS = 14;                             // Number of daisy-chained shift registers
const uint8_t PINS_PER_CHIP = 8;                                          // Number pins provided by each shift register chip
const uint8_t NUMBER_OF_PINS = NUMBER_OF_SHIFT_REGISTERS * PINS_PER_CHIP; // Number of pins available in all daisy-chained shift registers

const uint8_t NUMBER_OF_ENCODERS = 1;

const uint8_t DEFAULT_PULSE_LENGTH = 3;
const uint8_t ENCODER_CHANGE_MODE_HOLD_LENGTH = 5;

const uint8_t CYCLES_MAX = 100;

static uint8_t cyclesCounter = 0;

enum PinState : uint8_t
{
    STAYED_OPEN,
    STAYED_CLOSE,
    OPENED,
    CLOSED
};

static PinState currentPinsStates[NUMBER_OF_PINS + 1];

static uint8_t pulsedPinsReleaseTimestamps[NUMBER_OF_PINS + 1];

byte shiftRegistersState[NUMBER_OF_SHIFT_REGISTERS];
byte shiftRegistersPreviousState[NUMBER_OF_SHIFT_REGISTERS];

enum InputType : uint8_t
{
    NORMAL_INPUT,                        // Press the button on pin closed, release the button on pin opened
    REVERSED_INPUT,                      // Press the button on pin opened, release the button on pin closed
    NORMAL_PULSE_INPUT,                  // Pulse the button on pin closed
    REVERSED_PULSE_INPUT,                // Pulse the button on pin opened
    NORMAL_AND_REVERSED_PULSE_INPUT,     // Pulse the button on pin opened and closed
    IGNORE_INPUT,                        // Ignore pin being open or closed
    CHANGE_ENCODER_MODE_ON_PRESS_HOLD,   // Change encoder mode when button is released for a ENCODER_CHANGE_MODE_HOLD_LENGTH number of cycles
    CHANGE_ENCODER_MODE_ON_RELEASE_HOLD, // Change encoder mode when button is pressed for a ENCODER_CHANGE_MODE_HOLD_LENGTH number of cycles
};

class InputMapping
{
public:
    InputType _inputType;
    uint8_t _buttonIndex;

    InputMapping()
[... 479 lines cut ...]
                Joystick.setButton(inputMappings[pinIndex]._buttonIndex, 0);
            }
            else if (inputMappings[pinIndex]._inputType == CHANGE_ENCODER_MODE_ON_RELEASE_HOLD)
            {
                for (uint8_t encoderIndex = 0; encoderIndex < NUMBER_OF_ENCODERS; encoderIndex++)
                    if (encoders[encoderIndex]._switch == pinIndex)
                    {
                        encoders[encoderIndex].releaseOngoingPulseInput();
                        encoders[encoderIndex].changeEncoderMode();
                        encoders[encoderIndex].allocateInputMappings();
                    }
            }
        }

        if (cyclesCounter == CYCLES_MAX && pulsedPinsReleaseTimestamps[pinIndex] > 0)
            pulsedPinsReleaseTimestamps[pinIndex] = pulsedPinsReleaseTimestamps[pinIndex] - 1;
    }
}
```

## Wiring Details
Any pin wiring details that may be relevant.

## Additional context
Add any other context about the problem here.

MHeironimus, 2020-08-20:
What program are you using that indicates the joystick's type is "Driving"?

MaciejNiklasinski, 2020-08-20:
Ages ago I wrote in C# a vJoy feeder/remapper in which I used SharpDx.DirectInput package to detect all the connected 
HIDs. I have every reason to believe that it detects the device type correctly because it detects my Thrustmaster Warthog as "Joystick" and my Logitech G29 as "Driving".

[image]
[image]
[image]
[image]

MHeironimus, 2020-08-20:
Thanks for the extra information. I'll have to investigate further to see if I can figure out what DirectInput is using to determine the joystick type.

Note to myself: Looks like it may be related to this: https://docs.microsoft.com/en-us/windows/win32/api/dinputd/ns-dinputd-dijoytypeinfo?redirectedfrom=MSDN