MHeironimus/ArduinoJoystickLibrary #164: In Game Controllers everything works fine but erratic in games

EloyFarina, 2020-09-04:
Board: Arduino Leonardo R3 (clone but looks identical to the original)
SO: Windows 10 x64
Joystick: SEGA Genesis MK-1470 6-Button Turbo Controller (clone)
Code used: "GamepadExample.ino" (without modify)

In Game Controllers the Arduino Leonardo joystick works as expected. D-Pad and the B button works correctly, however in all games that I've tested, it moves to the left continuously, even if I touch B button.

Not sure if this behavior is related to the code. Any help will be appreciated.

[image]

Regards

EloyFarina, 2020-09-05:
I fixed the issue with x360ce software, creating a virtual XBOX 360 joystick, remapping D-PAD and the button, and hiding the Arduino Leonardo in Game Controllers panel with this same software.

MHeironimus, 2020-09-05:
Glad you were about to get your issue resolved.

closed 2020-09-05 by MHeironimus