MHeironimus/ArduinoJoystickLibrary #166: Weird flickering problem with analog input

Evilbolzen, 2020-09-29:
Hi there,

after facing some serious problems with the analog input ("flickering" signal) I used to work digital only (50+ Buttons thanks to shift-registers, pain to wire up but looks nice), however, now I am in need for "real" analog input signals and facing my problem again. 

So, reading an analog signal itself works just fine (using 10kOhm linear fader or potentiometer as well) and I was also able to create a visible joystick device with several different brand pro-micros and leonardos. 

Description of my problem:
[image]

Pic1: Inside the testing window everything looks good, the bar is moving according to the potentiometer movement. But, no matter if games, programs or my self written python scripts the signal is somehow "misunderstood". Every applications says the input is jumping between 0% up to several hundret percent. (Tested now 6 different vendors of pro micro, 3 different vendors of leonardo, 3 self-build Windows 10 PCs, 2 prebuild systems aka Dell, several Win7/8 Laptops, about 20 different USB cables, 30+ different potentiometer/fader, several breadboards, soldering, wiretypes etc.).

When having a closer look at the joystick settings in windows I recognized something fishy inside the calibration menu:

[image]

[image]

Pic 2 & 3: 
As you can see, I put the analog input on the X-Rotation and the "raw" value of it remains constant, however the blue bar above with the percentage is literally flickering around and showing all sorts of numbers. This happens with every axis and every analog pin I could get my hands on. This behaviour is exactly what I am seeing inside my different applications, 

So I reduced the example-code and even looked for answers with MS-tech support, nothing helped (and on my main-pc other, commercial analog devices work just fine).  I mean, the signal as number is read correctly, but why is the percentage flickering like it has a seizure?!? Doesn't make sense to me.

So maybe someone now a new approach to my problem or has an idea. My "minimal" Code is shown below. Didn't remove everything unneccessary as I am still working on a solution right now.
Thanks in advance

`
#include <Keypad.h>
#include <Joystick.h>

#define ENABLE_PULLUPS

#define NUMBUTTONS 16
#define NUMROWS 4
#define NUMCOLS 4

int X1 = A0;
int Y1 = A0; //A1
int rudder = A0; //A2
int throttle = A0; //A3

byte buttons[NUMROWS][NUMCOLS] = {
  {0, 1, 2, 3},
  {4, 5, 6, 7},
  {8, 9, 10, 11},
  {12, 13, 14, 15},
};

byte rowPins[NUMROWS] = {6, 7, 8, 9};
byte colPins[NUMCOLS] = {2, 3, 4, 5};

Keypad buttbx = Keypad( makeKeymap(buttons), rowPins, colPins, NUMROWS, NUMCOLS);

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_GAMEPAD, 0, 0,
  false, false, false,              // X and Y, Z Axis
  true, false, false,               // Rx, Ry, Rz
  false, false,                     // rudder, throttle
  false, false, false);             // accelerator, brake, steering

void setup() {
  Joystick.begin();
  Joystick.setRxAxisRange(0, 1023);
}

void JButtonStates() {
  Joystick.setRxAxis(analogRead(X1));
}

void loop() {
  JButtonStates();
  delay(50);
  CheckAllButtons();
}

void CheckAllButtons(void){ 
  if (buttbx.getKeys()){
    for (int i = 0; i < LIST_MAX; i++){
      if ( buttbx.key[i].stateChanged )  {
        switch (buttbx.key[i].kstate)  {
          case PRESSED:
          case HOLD:
          Joystick.setButton(buttbx.key[i].kchar, 1);
          break;
          case RELEASED:
          case IDLE:
          Joystick.setButton(buttbx.key[i].kchar, 0);
          break;
       }
      }
    }
  }
}
`
PS: I also tried JOYSTICK_TYPE_JOYSTICK and also a bunch of different ranges for the axis, however went back to the original 0-1023

byteborg, 2020-10-10:
I got this too, in the calibration dialog of Windows. A quick check with `jstest` on Linux shows no erratic behaviour. I think that the calibration dialog in Windows is b0rken. When using my simracing pedal set, the inputs all appear to be ok and not jittery at all.

Evilbolzen, 2020-10-12:
@byteborg 
Interesting, will have a look on linux too within the next couple of days. Was still busy working on windows and tried a few things. First I tried using a teensy, I know, technically wrong github for this but it is working. I guess the microcontroller overhaul from the teensy guys was quite good. But since these boards are freaking expensive not really an option (in total I'll need analog input for 12 different computers).

But, using the teensy got me thinking, I believe there is a mapping issue within windows (as you said, it works just fine under linux). So I modified my code and tried to map the raw values (0-1023) to the actual windows standard of -32767- +32767) and had a look at the serial output of the leonardo. (not configured as a joystick yet, just a quick value check). And now guess what, I get jittery values in my serial monitor, just like in the calibration menu.
So I guess there is a problem with the 16bit integer values? Wouldn't be surprised if windows is getting the numbers wrong, not the first time I am experiencing something like that.

Will report after I made some further tests

LordPato, 2020-11-22:
Try to add a 103 cap to ground to fix the flickering issue on analog signal

mfilia, 2020-12-08:
@Evilbolzen Did you find the source of the problem in the end? I'm curious as I have some 10k pots arriving. Thanks

Evilbolzen, 2020-12-08:
> @Evilbolzen Did you find the source of the problem in the end? I'm curious as I have some 10k pots arriving. Thanks

Hello there,

partially yes. Why partially? Because I managed to kill one of my potentiometers in the process and can't confirm a full solved on the problem. 
First things first: The flickering in the Windows menu can be ignored as far as I can tell now. Nearly any application uses the raw values provided by the arduino and not the "translated" value from Windows. And even if not, in most cases you can switch the input to the right values by choosing the arduino as game controller. So far only 1 self-written programm in my repertoire is still using the windows values, but it is only a matter of time until I solved that issue too (btw: it's an analogue brightness/color control for ImageJ/FiJi)

What you have to look for: hook up every potentiometer one by one, it is crucial to wire them absolute identically (slider positioned in the middle when booting up!). However, on ultra-cheap potentiometers you might experience major difference in quality and even in the official pin-out (that's why I killed my potentiometers), so check every pinout and hook them up step by step. If you have bad quality your raw-signal might flicker, as "LordPato" suggested use a cap to ground, helped a lot.

I am still working on the perfect solution without any weird issues, currently I am testing some diodes to prevent some other side effects when working with 5 potentiometers. Also working on an analog multiplexer implementation (CD4051) as I want to reduce the number of arduinos. Made this work for digital input already (90+ Buttons working like a charm :D )

ttait-vantim, 2020-12-10:
Yes the windows calibration GUI on some axis appears busted. X,Y seems fine but throttle flickers wildly, even on my Thrustmaster. You can always print the ADC values from the arduino at the same time to see what the readings were.

I used 10K pots w/o issue in FS2020. A 0.1uF cap from VCC to GND is always a good idea near the pot inputs, a 0.01uF cap on the center tap to GND will smooth any pot glitches. If you go higher than 10K, the source impedance feeding the ADC is too high and will cause some inaccuracy, but probably not an issue for this application.

I buy all my components from authorized distributors (digikey, mouser, newark, allied) tho, never ali express - fraud, counterfeit and quality rejects are rampant there. Bourns 10K pots are ~$1.50 in low qty. You can also get adafruit/sparkfun/seeed boards from them so it's largely one-stop shopping with same day fulfillment, 2day shipping is $10. All with proper ESD handling/packaging.

Tim

mastersta, 2021-06-07:
I had a similar issue with very "noisy" joystick values in my simulator and the W10 calibration tool, and I completely fixed it by changing the initAutoSend to `true` where I was previously using `false` in combination with `joystick.sendState()`. This completely removed the jittery behavior of my analog axes.