MHeironimus/ArduinoJoystickLibrary #167: Failed to compile JoystickTest sample sketch

seccpur, 2020-10-02:
The sample JoystickTest sketch failed to compile. On the Windows's Device Manager, the board is detected as "Arduino Leonardo"

**Hardware:** Arduino Leonardo
**Arduino IDE:** 1.8.13
**Sample Example:** Joysticktest.ino
**OS:** Windows 10

```
#define USBCON   // to disable unsupported hardware error
#include "Joystick.h"
```

This is the error code on Arduino IDE:

```
> Arduino: 1.8.13 (Windows 10), Board: "Arduino Uno"
> 
> In file included from C:\Users\Encoded\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/DynamicHID/DynamicHID.h:37:0,
> 
>                  from C:\Users\Encoded\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/Joystick.h:24,
> 
>                  from D:\Justin Projects\Arduino Papa's Lessons\JoystickTest\JoystickTest.ino:13:
> 
> C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/PluggableUSB.h:35:22: error: 'USBSetup' has not been declared
> 
>    virtual bool setup(USBSetup& setup) = 0;
> 
>                       ^~~~~~~~
> 
> C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/PluggableUSB.h:37:29: error: 'USBSetup' has not been declared
> 
>    virtual int getDescriptor(USBSetup& setup) = 0;
> 
>                              ^~~~~~~~
> 
> C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/PluggableUSB.h:57:21: error: 'USBSetup' has not been declared
> 
>    int getDescriptor(USBSetup& setup);
> 
>                      ^~~~~~~~
> 
> C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/PluggableUSB.h:58:14: error: 'USBSetup' has not been declared
> 
>    bool setup(USBSetup& setup);
> 
>               ^~~~~~~~
> 
> In file included from C:\Users\Encoded\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/Joystick.h:24:0,
> 
>                  from D:\Justin Projects\Arduino Papa's Lessons\JoystickTest\JoystickTest.ino:13:
> 
> C:\Users\Encoded\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/DynamicHID/DynamicHID.h:91:3: error: 'InterfaceDescriptor' does not name a type
> 
>    InterfaceDescriptor hid;
> 
>    ^~~~~~~~~~~~~~~~~~~
> 
> C:\Users\Encoded\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/DynamicHID/DynamicHID.h:93:3: error: 'EndpointDescriptor' does not name a type
> 
>    EndpointDescriptor  in;
> 
>    ^~~~~~~~~~~~~~~~~~
> 
> C:\Users\Encoded\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/DynamicHID/DynamicHID.h:117:21: error: 'USBSetup' has not been declared
> 
>    int getDescriptor(USBSetup& setup);
> 
>                      ^~~~~~~~
> 
> C:\Users\Encoded\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/DynamicHID/DynamicHID.h:118:14: error: 'USBSetup' has not been declared
> 
>    bool setup(USBSetup& setup);
> 
>               ^~~~~~~~
> 
> exit status 1
> 
> Error compiling for board Arduino Uno.
> 
> 
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
>

gdsports, 2020-10-02:
The compile fails because the board type is set to Uno.

`> Error compiling for board Arduino Uno.`

closed 2020-10-02 by MHeironimus

seccpur, 2020-10-03:
Works like charm. Thanks.