MHeironimus/ArduinoJoystickLibrary #168: Button issue

Blackman6560, 2020-10-03:
## Description of Issue
I'm trying to build a controller for Microsoft Flight Simulator X. I have buttons and potentiometers. My issue is that if I press and hold a button down the simulator just gets a short pulse signal instead of a long one. 
For example if I bind braking to a button and hold it down, it will just brake for a moment.
But the USB Game Controller panel in Windows can see that the button is still pressed down. (So it gets the whole pulse.)
Maybe the problem is with my simulator, but if I plug in a PS3 controller it works properly, with the whole signal length.
Have you got any idea how could I fix it?

## Technical Details
- Arduino Leonardo
- Windows 10
- Microsoft Flight Simulator X (FSX)

## Wiring Details
Arduino's internal pullup resistors used at the buttons.
The button pulls the digital pin down to GND.

##Code
[FSXController.zip](https://github.com/MHeironimus/ArduinoJoystickLibrary/files/5322294/FSXController.zip)

theblackhawk76, 2020-11-02:
Which code are you using? Try this one:

[AMSTODIU CODE BUTTON BOX](https://github.com/AM-STUDIO/USB-5-AXIS-CONTROLLER)

[AMSTUDIO VIDEO](https://www.youtube.com/watch?v=Z7Sc4MJ8RPM)

for me it worked perfectly and sends a constant signal.

Regarding the potentiometer I once created my own code:

```
//Code on the basis of: https://github.com/AM-STUDIO/USB-5-AXIS-CONTROLLER
//and: https://github.com/AM-STUDIO/USB-5-AXIS-CONTROLLER
//by AMSTUDIO: https://www.youtube.com/watch?v=Z7Sc4MJ8RPM

//Created by The Balck Hawk 76: https://github.com/theblackhawk76

//Arduino Leonardo
// six Axis on the six individual A-Pins, the GND pin and the 5v pin.
//25 buttons from pin: 12,11,10,9,8 for the 5 rows and pin 6,5,4,3,2 for the five columns of the matrix.
//1 rotary encoder at pin 0 and 1 and GND.
//AMSTUDIO Video might help: https://www.youtube.com/watch?v=Z7Sc4MJ8RPM

#include <Keypad.h>
#include <Joystick.h>

// Create Joystick
Joystick_ Joystick;

int zAxis_ = 0; 
int RxAxis_ = 0;                    
int RyAxis_ = 0;                                       //six axis
int RzAxis_ = 0;          
int Throttle_ = 0;
int xAxis_ = 0;         

const bool initAutoSendState = true; 

#define ENABLE_PULLUPS
#define NUMROTARIES 1
#define NUMBUTTONS 24
#define NUMROWS 5
#define NUMCOLS 5

//define the symbols on the buttons of the keypads
byte buttons[NUMROWS][NUMCOLS] = {
  {0,1,2,3,4,},
  {5,6,7,8,9,},
  {10,11,12,13,14,},
  {15,16,17,18,19,},
  {20,21,22,23,},
};

struct rotariesdef {
  byte pin1;
  byte pin2;
  int ccwchar;
  int cwchar;
  volatile unsigned char state;
};

rotariesdef rotaries[NUMROTARIES] {
  {0,1,24,25,0},
  
  
};

#define DIR_CCW 0x10
#define DIR_CW 0x20
#define R_START 0x0
[... 124 lines cut ...]

unsigned char rotary_process(int _i) {
   unsigned char pinstate = (digitalRead(rotaries[_i].pin2) << 1) | digitalRead(rotaries[_i].pin1);
  rotaries[_i].state = ttable[rotaries[_i].state & 0xf][pinstate];
  return (rotaries[_i].state & 0x30);
}

void CheckAllEncoders(void) {
  for (int i=0;i<NUMROTARIES;i++) {
    unsigned char result = rotary_process(i);
    if (result == DIR_CCW) {
      Joystick.setButton(rotaries[i].ccwchar, 1); delay(50); Joystick.setButton(rotaries[i].ccwchar, 0);
    };
    if (result == DIR_CW) {
      Joystick.setButton(rotaries[i].cwchar, 1); delay(50); Joystick.setButton(rotaries[i].cwchar, 0);
    };
  }
}
```

I used an arduino leonardo. The six Axis are on the six individual A-Pins, the GND pin and the 5v pin.
The 25 buttons are from pin: 12,11,10,9,8 for the 5 rows and pin 6,5,4,3,2 for the five columns of the matrix as shown in the video.
You can connect a rotary encoder at pin 0 and 1 and GND.

I hoped I was able to help!

theblackhawk76, 2021-01-26:
Try using MMJOY2, it is much easier to use with many more buttons because it can read rotary encoders differently. However, you are limited to 6 encoders and 8 axes...

PS: Don't forget the diodes!!

VkTheProgrammer17, 2021-02-07:
you could use x360ce

VkTheProgrammer17, 2021-02-07:
it would recognize the xbox controller allowing you for better detection
btw, are you using breadboard? if yes then it might just be fumbling a little