MHeironimus/ArduinoJoystickLibrary #170: Rename Buttons

ASionville, 2020-10-21:
Hi everyone !

Is there a way to rename the buttons ?

I mean that rather than having Button0, Button1 ... in game, would it be possible to give a display name for each button ?
An example for this is the A, X, Y and B on Xbox Controllers

Thanks !

ASionville

markus-i, 2020-10-25:
As far as I know (i.e., unless there's a way doing this using the HID descriptor), that would be an issue of your host system's USB driver and/or application. It works for the XBox controller with Windows because there are only a few buttons - and both are made by the same company ;-).

MHeironimus, 2020-10-25:
I agree. My understanding this is a feature of the driver on the host computer, not the USB device.

closed 2020-10-25 by MHeironimus