MHeironimus/ArduinoJoystickLibrary #171: Axis Limit on Windows?

thomasnield, 2020-11-01:
Just a question. Does this really support more than eight axes on Windows? Someone told me recently that Windows cannot easily support more than 8 axes on a single controller. It would be helpful if I can get 9 to 11 axes for rudder, XY joystick, XY joystick #2,  and the rest for sliders and knobs... all as a single USB device.

If this is the case I may ditch Teensy for a Micro, just to use this great-looking library.

theblackhawk76, 2020-11-02:
https://www.microchip.com/forums/m501540.aspx

My gamecontroller in windows shows five axis as sliders and 2 as joystick which is the maxiumum I assume.

You need A-Pins for axis if I recall correctly and a leonardo for example has only 6....

The library itslef has these axis possibilities (Ignore the no's):

  true, true, false,     // X and Y, but no Z Axis
  false, false, false,   // No Rx, Ry, or Rz
  false, false,          // No rudder or throttle
  false, false, false);  // No accelerator, brake, or steering

https://github.com/MHeironimus/ArduinoJoystickLibrary/blob/master/examples/GamepadExample/GamepadExample.ino

I hope I was able to answer your question!

thomasnield, 2020-11-02:
@theblackhawk76 thank you, I think so. I do want to stress test this myself at some point, and need to find a compatible board with enough analog pins that will allow this (Teensy 3.6 does not seem to be supported). 

On another forum someone reported they used all 11 axes in this library in a random value test. I'll give it a try.

closed 2020-11-02 by thomasnield