MHeironimus/ArduinoJoystickLibrary #174: Tiller Wheel with Rotary encoder / potentiometer

V3nom536, 2020-11-17:
I was projecting this tiller wheel replica and came up that the best solution is by using a rotary encoder or a potentiometer, which one should i use and then can i use them with this library?
Im really new to arduino and stuff like this so im sorry for any "stupid" question i made / i'll make.
I red that attaching the central pin of a potentiometer to an input pin would of course make the board read the values generated by the pot but can i use this lib for translating the analogic inputs into "movements" in the virtual joystick of w10?
Thanks for further replies and again, sorry for any stupid question if made :)

MHeironimus, 2020-11-19:
This library is not concerned with what or how you read values, it only focuses on taking values and converting them into something a computer will see as a Joystick.

V3nom536, 2020-11-19:
Ok perfect so, ill assign my pot to an axis and it should work fine. Thanks so much

closed 2020-11-19 by MHeironimus