MHeironimus/ArduinoJoystickLibrary #177: 8 axes win 10 fixed

batbsv, 2020-12-17:
tested on win 10 pro only

file Joystick.cpp
---------------old
		if (includeRudder == true) {
			// USAGE (Rudder)
			tempHidReportDescriptor[hidReportDescriptorSize++] = 0x09;
			tempHidReportDescriptor[hidReportDescriptorSize++] = 0xBA;
		}

---------------------------------------------------------
		if (includeRudder == true) {
			// USAGE (Rudder)
			tempHidReportDescriptor[hidReportDescriptorSize++] = 0x09;
			tempHidReportDescriptor[hidReportDescriptorSize++] = 0xBB; change BA to BB
		}

Slider2 available now

[image]

batbsv, 2020-12-17:
new Joystick.h

`/*
  Joystick.h

  Copyright (c) 2015-2017, Matthew Heironimus

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/

#ifndef JOYSTICK_h
#define JOYSTICK_h

#include "DynamicHID/DynamicHID.h"

#if ARDUINO < 10606
#error The Joystick library requires Arduino IDE 1.6.6 or greater. Please update your IDE.
#endif // ARDUINO < 10606

#if ARDUINO > 10606
#if !defined(USBCON)
#error The Joystick library can only be used with a USB MCU (e.g. Arduino Leonardo, Arduino Micro, etc.).
#endif // !defined(USBCON)
#endif // ARDUINO > 10606

#if !defined(_USING_DYNAMIC_HID)

#warning "Using legacy HID core (non pluggable)"

#else // !defined(_USING_DYNAMIC_HID)

//================================================================================
//  Joystick (Gamepad)

#define JOYSTICK_DEFAULT_REPORT_ID         0x03
#define JOYSTICK_DEFAULT_BUTTON_COUNT       128
#define JOYSTICK_DEFAULT_AXIS_MINIMUM         0
#define JOYSTICK_DEFAULT_AXIS_MAXIMUM      1023
#define JOYSTICK_DEFAULT_HATSWITCH_COUNT      2
#define JOYSTICK_HATSWITCH_COUNT_MAXIMUM      2
#define JOYSTICK_HATSWITCH_RELEASE           -1
#define JOYSTICK_TYPE_JOYSTICK             0x04
#define JOYSTICK_TYPE_GAMEPAD              0x05
#define JOYSTICK_TYPE_MULTI_AXIS           0x08

class Joystick_
{
private:

    // Joystick State
	int16_t	                 _xAxis;
	int16_t	                 _yAxis;
	int16_t	                 _zAxis;
	int16_t	                 _xAxisRotation;
	int16_t	                 _yAxisRotation;
	int16_t	                 _zAxisRotation;
	int16_t                  _slider1;
	int16_t                  _slider2;
[... 85 lines cut ...]
	}
	inline void setSlider2Range(int16_t minimum, int16_t maximum)
	{
		_slider2min = minimum;
		_slider2max = maximum;
	}
	// Set Axis Values
	void setXAxis(int16_t value);
	void setYAxis(int16_t value);
	void setZAxis(int16_t value);
	void setRxAxis(int16_t value);
	void setRyAxis(int16_t value);
	void setRzAxis(int16_t value);
	void setSlider1(int16_t value);
	void setSlider2(int16_t value);
	void setButton(uint8_t button, uint8_t value);
	void pressButton(uint8_t button);
	void releaseButton(uint8_t button);
	void setHatSwitch(int8_t hatSwitch, int16_t value);
	void sendState();
};

#endif // !defined(_USING_DYNAMIC_HID)
#endif // JOYSTICK_h
`

batbsv, 2020-12-17:
new 

------------------------------
/*
  Joystick.cpp

  Copyright (c) 2015-2017, Matthew Heironimus

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/

#include "Joystick.h"

#if defined(_USING_DYNAMIC_HID)

#define JOYSTICK_REPORT_ID_INDEX 7
#define JOYSTICK_AXIS_MINIMUM -32767
#define JOYSTICK_AXIS_MAXIMUM 32767

#define JOYSTICK_INCLUDE_X_AXIS  B00000001
#define JOYSTICK_INCLUDE_Y_AXIS  B00000010
#define JOYSTICK_INCLUDE_Z_AXIS  B00000100
#define JOYSTICK_INCLUDE_RX_AXIS B00001000
#define JOYSTICK_INCLUDE_RY_AXIS B00010000
#define JOYSTICK_INCLUDE_RZ_AXIS B00100000
#define JOYSTICK_INCLUDE_Slider1 B01000000
#define JOYSTICK_INCLUDE_Slider2 B10000000

Joystick_::Joystick_(
	uint8_t hidReportId,
	uint8_t joystickType,
    uint8_t buttonCount,
	uint8_t hatSwitchCount,
	bool includeXAxis,
	bool includeYAxis,
	bool includeZAxis,
	bool includeRxAxis,
	bool includeRyAxis,
	bool includeRzAxis,
	bool includeslider1,
	bool includeslider2	)
{
    // Set the USB HID Report ID
    _hidReportId = hidReportId;

    // Save Joystick Settings
    _buttonCount = buttonCount;
	_hatSwitchCount = hatSwitchCount;
	_includeAxisFlags = 0;
	_includeAxisFlags |= (includeXAxis ? JOYSTICK_INCLUDE_X_AXIS : 0);
	_includeAxisFlags |= (includeYAxis ? JOYSTICK_INCLUDE_Y_AXIS : 0);
	_includeAxisFlags |= (includeZAxis ? JOYSTICK_INCLUDE_Z_AXIS : 0);
	_includeAxisFlags |= (includeRxAxis ? JOYSTICK_INCLUDE_RX_AXIS : 0);
	_includeAxisFlags |= (includeRyAxis ? JOYSTICK_INCLUDE_RY_AXIS : 0);
	_includeAxisFlags |= (includeRzAxis ? JOYSTICK_INCLUDE_RZ_AXIS : 0);
	
	_includeAxisFlags |= (includeslider1 ? JOYSTICK_INCLUDE_Slider1 : 0);	
	_includeAxisFlags |= (includeslider2 ? JOYSTICK_INCLUDE_Slider2 : 0);
[... 493 lines cut ...]
			{
				convertedHatSwitch[hatSwitchIndex] = (_hatSwitchValues[hatSwitchIndex] % 360) / 45;
			}			
		}

		// Pack hat-switch states into a single byte
		data[index++] = (convertedHatSwitch[1] << 4) | (B00001111 & convertedHatSwitch[0]);
	
	} // Hat Switches

	// Set Axis Values
	index += buildAndSetAxisValue(_includeAxisFlags & JOYSTICK_INCLUDE_X_AXIS, _xAxis, _xAxisMinimum, _xAxisMaximum, &(data[index]));
	index += buildAndSetAxisValue(_includeAxisFlags & JOYSTICK_INCLUDE_Y_AXIS, _yAxis, _yAxisMinimum, _yAxisMaximum, &(data[index]));
	index += buildAndSetAxisValue(_includeAxisFlags & JOYSTICK_INCLUDE_Z_AXIS, _zAxis, _zAxisMinimum, _zAxisMaximum, &(data[index]));
	index += buildAndSetAxisValue(_includeAxisFlags & JOYSTICK_INCLUDE_RX_AXIS, _xAxisRotation, _rxAxisMinimum, _rxAxisMaximum, &(data[index]));
	index += buildAndSetAxisValue(_includeAxisFlags & JOYSTICK_INCLUDE_RY_AXIS, _yAxisRotation, _ryAxisMinimum, _ryAxisMaximum, &(data[index]));
	index += buildAndSetAxisValue(_includeAxisFlags & JOYSTICK_INCLUDE_RZ_AXIS, _zAxisRotation, _rzAxisMinimum, _rzAxisMaximum, &(data[index]));
	
	index += buildAndSetAxisValue(_includeAxisFlags & JOYSTICK_INCLUDE_Slider2, _slider2, _slider2min, _slider2max, &(data[index]));
	index += buildAndSetAxisValue(_includeAxisFlags & JOYSTICK_INCLUDE_Slider1, _slider1, _slider1min, _slider1max, &(data[index]));
	
	DynamicHID().SendReport(_hidReportId, data, _hidReportSize);
}

#endif

ttait-vantim, 2020-12-17:
Is this related to Issue #44 ?

batbsv, 2020-12-18:
yes. additional, device visible in usb game control

[image]

theblackhawk76, 2021-01-26:
Maybe try using MMJOY2?

batbsv, 2021-01-26:
this issue fixed.work properly
mmjoy2 not have source code,  and project freezed.

thomasnield, 2021-02-12:
Thank you for this.

neilp2, 2021-02-28:
I can confirm this modification works, and makes the Rudder and Throttle present as two throttle inputs in windows Game Controllers panel:
[image]

I'm using this code to create the joystick with 8 analog inputs:
```
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, 
  JOYSTICK_TYPE_JOYSTICK, 24, 0,          // 24 buttons, no hat switch
  true, true, true, true, true, true,     // X, Y, Z, Xr, Yr, Zr
  true, true, false, false, false);       // Rud, Thr, Acc, Brake, Steer

```

This is providing a custom controller into Microsoft Flight Simulator 2020 (MSFS 2020). The two Throttles appear as 'slider' inputs Slider X and Slider Y.

[image]

Thank you @MHeironimus for the excellent library.

Aenonar, 2022-11-29:
No idea how or why this works but I can confirm that this fix works in DCS as well, thanks a lot!

[image]

jiangfei, 2023-10-13:
> 仅在win 10 pro上测试
> 
> 操纵杆_操纵杆(操纵杆_默认_报告_ID，操纵杆类型操纵杆，128，2，真，//x真，//y真，//z真，//rx真，//ry真，//rztrue，//rudder (Slider2)true，//throttle (Slider1)false，//accel假，//刹车假//转向 );
> 
> 文件操纵杆. cpp-旧的if (includeRudder == true) {//用法(方向舵)tempHidReportDescriptor[hidreportdescriptorsize++]= 0x 09；tempHidReportDescriptor[hidreportdescriptorsize++]= 0x ba； }
> 
> ```
> 	if (includeRudder == true) {
> 		// USAGE (Rudder)
> 		tempHidReportDescriptor[hidReportDescriptorSize++] = 0x09;
> 		tempHidReportDescriptor[hidReportDescriptorSize++] = 0xBB; change BA to BB
> 	}
> ```
> 
> 幻灯片2现已推出
> 
> [image]

After I changed the contents of the Jostick.cpp file, both throttles can be displayed in the game controller, but slider2 is not recognized in the DCS