MHeironimus/ArduinoJoystickLibrary #179: Joystick recognised but doesnt output any data

V3nom536, 2020-12-18:
Hello everyone, i just assembled a circuit that basically consist in a potentiometer and two buttons, the potentiometer should act as a x axis for my tiller wheel i should use with my flight sim. I connect the arduino to the pc, upload the sketch, gets recognised by w10 joystick interface but it doesnt seems to work, no changes on the axis and no input from the buttons. I connected a led to one of the buttons and the led properly work and turn on, also the pot generated proper values as i tested it with the serial monitor
Im currently using an Arduino Pro Micro board.
This is the sketch
`#include <Joystick.h> //import library

Joystick_ Joystick; //creates virtual joystick
int parkButton = 6; //park brake button
int LED = 4;
int stateLED = LOW;
int stateButton;
int previous = LOW;
long time = 0;
long debounce = 200;
//declared park brake led variables (UP)

int XAxis_ = 0; //pot axis declared

int brakeButton = 21; //declared pedal brake button to pin 21 (A3)

void setup() {
  pinMode(parkButton, INPUT);
  pinMode(LED, OUTPUT);  
  //defined pinmodes for led variables (UP)

  pinMode(brakeButton, INPUT);
}

void loop() {

  int XAxis_ = analogRead(A8);  //reads input from A8 pin
  XAxis_ = map(XAxis_, 0, 1023, 255, 0); //map the xaxis variable
  Joystick.setXAxis(XAxis_);  //move the axis in the virtual joystick
  //code related to main knob
  
  stateButton = digitalRead(parkButton);  
  if(stateButton == HIGH && previous == LOW && millis() - time > debounce) {
    if(stateLED == HIGH){
      stateLED = LOW; 
    } else {
       stateLED = HIGH; 
    }
    time = millis();
  }
  digitalWrite(LED, stateLED);
  previous == stateButton;
  //code related to park led (UP)

  if (digitalRead(6)== HIGH) //if the PARK button is pressed, HIGH value related to button + resistor config
  {Joystick.setButton(1, 1);} //set the button 1 to value 1 (hold press)
  else  //if the button is NOT PRESSED
  {Joystick.setButton(1, 0);} //set the button 1 to value 0 (release)
  //code related to park brake button

  if (digitalRead(21)== HIGH) //if the PEDAL button is pressed
  {Joystick.pressButton(2);}  //press the button 2
  else  //if the button 2 is NOT BEING PRESSED
  {Joystick.releaseButton(2);} //relase the button 2
  //code related to "maintain-to-activate" state pedal brake button
}`

Im really new to arduino so i think i misunderstood some concepts, sorry in advance for any stupid question i may ask or i asked.
Thanks everyone for support!

ttait-vantim, 2020-12-19:
How did you wire the pot up? What value is it? Is it linear or log (audio) curve?

Tim

V3nom536, 2020-12-22:
The values look perfect, set the range (0, 1023, 255, 0) in setup and everything seems fine but in the sim most of the pot range wasnt moving the tiller at all. It was nearly required to move full left or full right to give an input, after i set the response curve in the exact point the pot was starting to "generate input" then the calibration was perfect and i got full pot's range working

closed 2021-01-09 by MHeironimus

ConBeeUser, 2021-01-11:
> The values look perfect, set the range (0, 1023, 255, 0) in setup and everything seems fine but in the sim most of the pot range wasnt moving the tiller at all. It was nearly required to move full [...]

I bet this is relatet to using the map-function.
You map the range of 0-1023 to 255-0 which means that the scale is inverted and reduced to a quarter of its full length. I think you should try removing map or as I did first:
map(0, 1023, 0, 1023);

which of cource is useless (as it maps e.g. a read value of 750 to 750 but you can quickly revert it this way.

If your max read value of 1023 just moves the in game value by about a quarter of the available slider, this might be the reason.

Cheers Mat