MHeironimus/ArduinoJoystickLibrary #186: Why does this code "brick" the Arduino?

espenrs, 2021-01-12:
This extremely simple code that I thought would work as a poc basically bricks the Leonardo. I can get it back again by burning the bootloader via an Uno, but the code still bricks the Leonardo - each time:
```
#include "Joystick.h"

Joystick_ Joystick;

int analogPin = A0;
int value = 0;

void setup() {
  Joystick.setThrottleRange(0, 1023);
  Joystick.begin();
}

void loop() {
  value = analogRead(analogPin);
  Joystick.setThrottle(value);
}
```

MHeironimus, 2021-01-12:
I think you need a delay in your loop() function. You are probably sending too much data to the host PC too fast.

espenrs, 2021-01-14:
I'm getting ambiguous results. In short, a delay of 50 or 100 didn't work completely. It might be better, or it might be the same. I need to test more and see if a pattern emerges. It also might just be a faulty board. Unless someone else have experienced this issue it really might just be the board.

espenrs, 2021-02-14:
Pretty sure by now that my issue was a faulty USB cable.

closed 2021-02-14 by espenrs