MHeironimus/ArduinoJoystickLibrary #187: Joystick on M1 Mac

Ja-Hey-64, 2021-01-14:
I# been trying to make a sim Wheel to work with my M1 Mac Book but have'nt quite got it to work in game the axes show up but only have the input value assigned to them, that was present on start up. This means I can change the input value with reuploading my code but not with something like a potentiometer. For comparison I also testet the exact same code on an Intel Mac and it worked just fine. Maybe anybody has an Idea? I think this may be an issue with how the new Mac handle they IO but I am really not an expert when it comes to USB communication.

## Technical Details

- Arduino Board: Arduino pro micro clone 
- Host OS: Mac OS Big Sur
- Arduino IDE Version (e.g. 1.8.3):

## Sketch File that Reproduces Issue

~~~
Sketch file goes here (if applicable)
~~~

## Wiring Details
Any pin wiring details that may be relevant.

## Additional context
Add any other context about the problem here.

MHeironimus, 2022-03-30:
You might try updating to the latest version of the library (2.1.0) and see if you are still having the issue on the M1 Macs. I do not have an M1 Mac to test with.

Ja-Hey-64, 2022-03-31:
The issue seems to not be caused by the joystick library, but rather by the rosetta translation. I previously tested it with Euro Truck simulator 2 and the Farming Simulators 2017/19. In the recently released Farming Simulator 2022 the device works just as expected (this game runs natively). 
I am sorry for getting this wrong and am very thankful for this great library.

closed 2022-03-31 by Ja-Hey-64