MHeironimus/ArduinoJoystickLibrary #189: Naming the controller

jaygrovr, 2021-01-20:
**Describe your feature request**
It would be great if there was a way to put in a custom name for the controller, when it is detected by windows.  Currently, it just shows "Arduino Leonardo".  When using more than one arduino, its hard to tell which one is plugged in and working.  Having a name for it would help avoid that confusion.  I see that there is an ID you have to change for each device you are plugging in.

**Additional context**
Add any other context or screenshots about the feature request here.

VkTheProgrammer17, 2021-01-24:
watch this video -https://www.youtube.com/watch?v=hoCOq9Ngp44 from 20 minutes

jaygrovr, 2021-01-24:
This is excellent, but my hardware folder didnt exist, as far as I can tell.  What about the AVR board folder inside?  I dont have that either? I have multiple boards installed on my IDE..
Any thoughts?
jason

jaygrovr, 2021-01-24:
Ok i saw I had to download the files.  I still cant get the new board to show up in the IDE.  Second question, how can I make two entries for two different HID joysticks (i have two leonardos).
Thanks
Jason

jaygrovr, 2021-01-24:
Nevermind on the location, i found it.  Messy computer.  Still want to know how to create a second.  Do I add another entry to the boards.txt or do I add another boards.txt file with a different name and different contents?
Thanks
Jason

VkTheProgrammer17, 2021-01-24:
i think its the latter. i have not downloaded the files myself so i'm just guessing

VkTheProgrammer17, 2021-01-24:
the person from the video(which was very informative) said he was going to upload a video about making multiple ids

jaygrovr, 2021-01-24:
Ok. yea, now I cant get one joystick working..
Do all the types exist for all the axes?
For instance if i set type to JOYSTICK_TYPE_JOYSTICK , do all the analog sticks become available for use?  The 11 axes?  Or are they different for each stick type?
I keep on getting an error in windows on the USB device.  It uploads fine.  I have a 40ms delay at the end of theloop, but it never shows up.  Even setting the board back to leonardo, it still doesnt show up.
Jason

VkTheProgrammer17, 2021-01-24:
if you type only JOYSTICK_TYPE_JOYSTICK, it thinks that there are 32 buttons, 2 hat switches and all axes. maybe you can watch the whole video to understand the library better. i am also doing the same. btw, you can check out my repository-https://github.com/VkTheProgrammer17/gaming-steering-wheel

jaygrovr, 2021-01-24:
Yes i got it all to work fine.. I didnt know the limitations of the Rx Ry and Rz axes.. and that you have to enable all three or none.

VkTheProgrammer17, 2021-01-30:
you're welcome

closed 2021-02-28 by MHeironimus