MHeironimus/ArduinoJoystickLibrary #191: Can I use this library for two-way communication?

bert-laverman, 2021-01-28:
I have a set of buttons with built-in LEDs. Perfect for e.g. an MCP or such, where I push the buttons to engage the autopilot and have the led light up if the corresponding function actually engages. I found a nice C++ library that allows me to communicate with HID devices, and the Arduino Micro shows up properly. Detecting the buttons works fine, but now I need to be able to pass the led state from PC to Arduino, which means I need to send some kind of message and pick it up in the "`DynamicHID.cpp`" but has anyone worked out how to do this?

Cheers,
Bert Laverman

VkTheProgrammer17, 2021-02-07:
so do you want the LED to turn on when you push the button?

bert-laverman, 2021-02-07:
Yes and No: I do not want the Arduino to decide when the LEDs need to go on. That is up to what happens on the PC. So what I want is comparable to the CAPS-lock and such LEDs, where the PC sends a message (event? report?) to the Arduino telling it which LEDs must go on or off.

VkTheProgrammer17, 2021-03-28:
hey! i found something on youtube for what you need.https://www.youtube.com/watch?v=aMato4olzi8 you can wrap all of his code up in a function and use your push button like caps lock