MHeironimus/ArduinoJoystickLibrary #192: How to get encoder switches to work?

laserbeak, 2021-02-06:
Hi,
Orignially, I used this code to get encoder switches to work:
`
#include <Joystick.h>

#define CLK_PIN1  1
#define DATA_PIN1 0
#define KBUTTON1  2
#define CLK_PIN2  21
#define DATA_PIN2 20
#define KBUTTON2  19
#define BUTTON1 3
#define BUTTON2 4
#define BUTTON3 5
#define BUTTON4 6
#define BUTTON5 10
#define BUTTON6 16
#define BUTTON7 14
#define BUTTON8 15

Joystick_ Joystick;

#define JOYSTICK_TYPE_GAMEPAD
////////////////////////////////////////////////////
void setup() {
   pinMode(CLK_PIN,INPUT);
   pinMode(DATA_PIN,INPUT);
   pinMode(KBUTTON, INPUT_PULLUP);

   Serial.begin(9600);
   Serial.println("Rotary Encoder KY-040");
}

void loop() {
static uint16_t state=0,counter=0;

    delayMicroseconds(100); // Simulate doing somehing else as well.

    state=(state<<1) | digitalRead(CLK_PIN) | 0xe000;

    if (state==0xf000){
       state=0x0000;
       if(digitalRead(DATA_PIN))
         counter++;
       else
         counter--;
       Serial.println(counter);
    }

    if (digitalRead(BUTTON)==0) {

      delay(10);
      if (digitalRead(BUTTON)==0) {
          Serial.println("Next Detent");
          while(digitalRead(BUTTON)==0);
      }
   }
}`

How would I use your library to achieve something like this?

VkTheProgrammer17, 2021-02-07:
Encoder switches are just like normal 2-pin push button switches, so whatever code you have written for your switches, use that code for these encoder switches.

laserbeak, 2021-02-15:
I don't follow. Will including joystick.h make this code visible to the windows directInput  Game Controller manager?

closed 2021-02-16 by laserbeak

VkTheProgrammer17, 2021-02-17:
you are welcome