MHeironimus/ArduinoJoystickLibrary #194: Is it possible to have more sliders?

espenrs, 2021-02-19:
Is it possible to have more sliders like the Throttle? Or is this a limitation in the default Windows 10 controller drivers?

I'm creating a controller where I will have up to 7 sliders - 2d axes without x and y coordinates. I basically have a bunch of separate x axes. Is this possible to do or does this require custom Windows 10 drivers?

82ndab-Bravo17, 2021-02-26:
You can just use any of the axes for a slider (which is actually 1d), neither the library nor windows will care that you are not actually using them as part of a 2d or 3d joystick.

I think that the Windows limit is 8 axes per controller. I am not sure if there is a limit to the number of controllers.

espenrs, 2021-03-03:
Yes, it is possible to use any of the axes as a slider. But the appearance would be much better if they looked like a slider in the Game controller settings. I'm looking for any conformation that it is possible or the sad truth that the default drivers simply doesn't support anything other than the hard coded axes and sliders.

The same with the names. I would love to have sliders called Propeller and Mixture. But I suspect the names are also hard coded into the default drivers. Don't you love it when people decides everything for you so you don't have to be so innovative and can just stick with what's been there for decades?