MHeironimus/ArduinoJoystickLibrary #198: setXAxis and setYAxis not properly translating values to -32K/32K grid

Hay1tsme, 2021-04-07:
## Description of Issue
I was using [this](https://github.com/knuckleslee/RhythmCodes/tree/master/2E10B10LED_sdvx/leo) sketch to build a game controller, when I noticed the axis displaying funky behavior. Incrementing the value of the axis registered in windows as widely different places on the joystick as shown in both the windows controller configuration window, and in the game I was attempting to play. When the axis range is set from -25 to 24, the following are some examples of the values that windows interprets
| setYAxis value | value in windows |
|--------------- |------------------ |
| 4                    | 31300                 |
| 3                    | 24639                 |
| 2                    | 27330                 |
| 1                    | 28662                 |
| 0                    | 31343                 |

I confirmed the issue by using the simple sketch below, and got the same results for both axis.

## Technical Details

- Arduino Board (e.g. Arduino Leonardo): Arduino Micro
- Host OS (e.g. Windows 10): Windows 10 20H2
- Arduino IDE Version (e.g. 1.8.3): 1.8.13

## Sketch File that Reproduces Issue

~~~c
#include <Joystick.h>
Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID,JOYSTICK_TYPE_GAMEPAD, 10, 0,
 true, true, false, false, false, false, false, false, false, false, false);

//int encL=0, encR=0;
const int PULSE = 50;  //number of pulses per revolution of encoders 
byte EncPins[]    = {0, 1, 2, 3};
byte EncPinCount = sizeof(EncPins) / sizeof(EncPins[0]);

void setup() {
  Serial.begin(9600);
  Joystick.begin();
  Joystick.setXAxisRange(-PULSE/2, PULSE/2-1);
  Joystick.setYAxisRange(-PULSE/2, PULSE/2-1);

  for(int i=0;i<EncPinCount;i++) {
    pinMode(EncPins[i],INPUT_PULLUP);
  }
}

void loop() {
  for (int i = -PULSE/2; i < PULSE/2; i++) {
    Joystick.setXAxis(i);
    Serial.println(i);
    delay(500);
  }
  for (int i = -PULSE/2; i < PULSE/2; i++) {
    Joystick.setYAxis(i);
    Serial.println(i);
    delay(500);
  }
  

}
~~~

enghady-ahmed, 2021-09-17:
I have same problem i notice that seting range for the axial not change and that appears when i mark display vale in joystick calibration

MHeironimus, 2022-03-30:
Try updating to the latest version of the library (2.1.0) and see if you are still having this issue.