MHeironimus/ArduinoJoystickLibrary #203: Adding this line in setup causes updates to stop

paulwrath1223, 2021-05-04:
## Description of Issue
Adding these lines after "serial begin" in setup causes updates to stop, despite "set axis" being executed
_while (!Serial)
    delay(10);_
It is part of the example sketch for the mpu 6050 gyro.
[joystick INO.zip](https://github.com/MHeironimus/ArduinoJoystickLibrary/files/6418757/joystick.INO.zip)

## Technical Details
IDK I'm a noob
- Arduino Board Arduino Leonardo
- Host OS Windows 10
- Arduino IDE Version 1.8.13

## Sketch File that Reproduces Issue

~~~
#include "Joystick.h"

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_JOYSTICK, 4, 0, false, false, false, false, false, false, false, false, true, true, true); 

void setup() {
  Joystick.setAcceleratorRange(0, 1023);
  Joystick.setBrakeRange(0, 1023);
  Joystick.setSteeringRange(0, 720);
  Serial.begin(115200);
//  while (!Serial)                                                    // GITHUB ISSUE HERE
//    delay(10);                                                       // uncomment those two lines and the joysticks will not move. the rest of the code is hot garbage, but thats my fault.

  Joystick.begin();

}

void loop() 
{

  Joystick.setSteering(10);
  Joystick.setBrake(10);
  Joystick.setAccelerator(10); 

}
~~~

## Wiring Details
You can test this with just a board
## Additional context
this might be really obvious to everyone else, but it seems to me that if the loop is running, (which can be confirmed with a debug print), the lines should be executed regardless of if it waited in setup. I fixed the problem by just waiting 1 second every power up and removing the problem lines.

Pomax, 2022-10-10:
Can confirm that this indeed prevents the joystick library from every actually doing anything, even if the rest of the code runs just fine.

bernatgy, 2026-10-26:
Can also confirm