MHeironimus/ArduinoJoystickLibrary #208: stuck with adding 4051 multiplexers for lots of buttons

Chong-McBong, 2021-05-24:
hello, i managed to get a 6 axis joystick joystick working with your code, many thanks, its marvellous :)

i'm stuck trying to add some 4051 multiplexers to add more buttons (running out of pins on my arduino micro).
I'm using 4051s because i already have them, and pretty sure its possible.

can anyone help adding the multiplexer code to the joystick code, i'm not sure where to start
I plan to read 24 buttons with 3 multiplexers, with shared S0,S1,S2 pins, and an input from each to the arduino.

i have wiring figured out, chose some pins to use:  (these pins leave all the analog pins, so i can use more if i need them later)

2 - S0 to 3x 4051 chips
3 - S1 to 3x 4051 chips
5 - S2 to 3x 4051 chips
7 - input from 4051 number 1
11 - input from 4051 number 2
13 - input from 4051 number 3

-------

`//  USB Joystick with hall effect sensors & pots, for Arduino (pro) Micro
//  using this library: https://github.com/MHeironimus/ArduinoJoystickLibrary

//  seems to be a limit of max 7 analogs
//  working 04-05-2021
//  Gonad Corp. 2021
//------------------------------------------------------------
// 3 4051 multiplexers = 24 buttons

 
//change these to define which pins your hall effect sensors or potentiometers are connected.
//to change button connections, scroll down to loop()
#define X_PIN A0    // X thumbstick
#define Y_PIN A1    // Y thumbstick
#define Z_PIN A2    // Z pot
#define Rx_PIN A3   // accelerator / pull throttle
#define Ry_PIN A4   // brake   
#define Rz_PIN A5   // clutch
//#define T_PIN A6  // unused axis 7 (A6 shares with digital pin 4)

//#define           //8th axis? not allowed

#include <Joystick.h>

Joystick_ Joystick(0x04,JOYSTICK_TYPE_JOYSTICK,
  12, 0,                // Button Count, Hat Switch Count
  true, true, true,     // X, Y, Z 
  true, true, true,     // Rx, Ry, Rz
  false, false,         // rudder, throttle
  false, false, false); // accelerator, brake, steering

void setup() {
// Initialize Button Pins   (gears etc. 1,2,3,4,5,6, Reverse , range)
//  pinMode(2, INPUT_PULLUP);
//  pinMode(3, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);   //turn off if axis 7 is enabled on A6
//  pinMode(5, INPUT_PULLUP);
  pinMode(6, INPUT_PULLUP);
//  pinMode(7, INPUT_PULLUP);
  pinMode(8, INPUT_PULLUP);
  pinMode(9, INPUT_PULLUP);
  pinMode(10, INPUT_PULLUP);
 // pinMode(11, INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
//  pinMode(13, INPUT_PULLUP);
 
  // Initialize Joystick Library
  Joystick.begin(false); //false = dont send automatically. We will sendState() at the end of the loop
  Joystick.setXAxisRange(1023, 0);  //(change 0,1023 to 1023, 0 to reverse)
  Joystick.setYAxisRange(1023, 0);
  Joystick.setZAxisRange(0, 1023);   
  Joystick.setRxAxisRange(1023, 0);  //accelerator & pull throttle via switch
  Joystick.setRyAxisRange(1023, 0);  //brake
  Joystick.setRzAxisRange(1023, 0);  //clutch        
  //Joystick.setThrottleRange(0, 1023);
}
 
 
void loop() {
  //read buttons. Change pins and button numbers here, if you want to have different number connected to different pin
  Joystick.setButton(0, !digitalRead(2)); //pin 2 LOW means button 0 PRESSED
  Joystick.setButton(1, !digitalRead(3)); //etc.
  Joystick.setButton(2, !digitalRead(4));
  Joystick.setButton(3, !digitalRead(5)); 
  Joystick.setButton(4, !digitalRead(6));
  Joystick.setButton(5, !digitalRead(7));
  Joystick.setButton(6, !digitalRead(8));
  Joystick.setButton(7, !digitalRead(9));
  Joystick.setButton(8, !digitalRead(10));
  Joystick.setButton(9, !digitalRead(11));
  Joystick.setButton(10, !digitalRead(12));
  Joystick.setButton(11, !digitalRead(13));
//Joystick.setButton(12,    read correct multiplexer jobby here  
//Joystick.setButton(13,     repeat
//Joystick.setButton(14, 
//Joystick.setButton(15, 
//Joystick.setButton(16, 
//Joystick.setButton(17,      
//Joystick.setButton(18, 
//Joystick.setButton(19, 
//Joystick.setButton(20, 
//Joystick.setButton(21, 
//Joystick.setButton(22,      
//Joystick.setButton(23, 
//Joystick.setButton(24, 
//Joystick.setButton(25, 
//Joystick.setButton(26, 
//Joystick.setButton(27,      
//Joystick.setButton(28, 
//Joystick.setButton(29, 
//Joystick.setButton(30, 
//Joystick.setButton(31, 

  //read analog axes
  int value = analogRead(X_PIN);
  Joystick.setXAxis(value);
  value = analogRead(Y_PIN);
  Joystick.setYAxis(value);
  value = analogRead(Z_PIN);  
  Joystick.setZAxis(value);
    
  value = analogRead(Rx_PIN);
  Joystick.setRxAxis(value);
  value = analogRead(Ry_PIN);
  Joystick.setRyAxis(value);
  value = analogRead(Rz_PIN);
  Joystick.setRzAxis(value);
    
  //value = analogRead(T_PIN);
  //Joystick.setThrottle(value);
 
  Joystick.sendState();
  delay(10);
}`

Gorgok, 2021-06-05:
I don't use those, but similar CD74HC4067M96 chips (16 channels, 1 input, 4 address pins). Basically what my code does is read the address (0-15) of the input with bitRead and makes the 4 address pins high/low to match that number in binary, then checks all the mux chips (since the address pins are shared) at that address. It does mean with this code the inputs will bounce back and forth from one mux chip to the other, instead of reading them all in a row from each chip.

#define MUX 2
#define INPUTS 16

// Mux pins
#define A 15
#define B 14
#define C 5
#define D 7

const int inputPins[MUX] = {16, 0};

...

//Joystick button number, reset back to 0 before loop
int i = 0;

// Loop through addresses
for (int address = 0; address < INPUTS; ++address){
    digitalWrite(A, bitRead(address, 0));
    digitalWrite(B, bitRead(address, 1)); 
    digitalWrite(C, bitRead(address, 2));
    digitalWrite(D, bitRead(address, 3));

    // Cycle through each input pin
      for (int mux = 0; mux < MUX; ++mux){
        Joystick.setButton(i++, !digitalRead(inputPins[mux]);
      }
   }

CDRXavier, 2021-08-20:
See [datasheet](https://www.ti.com/lit/ds/symlink/cd4051b.pdf)