MHeironimus/ArduinoJoystickLibrary #212: Change the device's serial ID

sonik-br, 2021-07-22:
**Describe your feature request**
Some systems differentiate arduino projects with same PID/VID by using the serial too.
It's possible to edit the serial info per sketch without editing the IDE/SDK files.

@MickGyver have done it on some projects like this one: [SegaTwoControllersUSB](/MiSTer-devel/Retro-Controllers-USB-MiSTer/tree/master/SegaTwoControllersUSB)

I would love to have this feature implemented in this lib but I don't have the skill to do it.

MHeironimus, 2021-07-22:
I think there is an explanation of how to do this at the end of the following video: https://www.youtube.com/watch?v=hoCOq9Ngp44 around 19:20.

sonik-br, 2021-07-22:
Yeah that's one way to do it. But would be a good option to edit it from the sketch file without having to edit the arduino ide files.

I see that the lib creates a serial based on the descriptorSize. It is an elegant solution to differentiate sketches with different descriptors.

I changed it to a custom string inside the method DynamicHID_::getShortName(char *name). And it worked.
Would like to have a way to set this custom string on the constructor of the joystick lib.

CDRXavier, 2021-08-19:
> I changed it to a custom string inside the method DynamicHID_::getShortName(char *name). And it worked.
Would like to have a way to set this custom string on the constructor of the joystick lib.

The upside of the library is that it tried to avoid all the nasty bits of USB (by not including code for "Arduino core"), but the downside is that in this way, the Arduino you use still appear to the computer as a Arduino (and not, say, a "sonik-br" brand joystick)
I had tried to "start from scratch", but it just keep getting deep. Like, very deep.
Deep enough that you can write USB code for a fresh 32U4 out of the factory (stuff written with Assembly).

sonik-br, 2021-08-20:
The idea is not to change the "name" that appears on the OS. But to change the "serial" that goes along with the VID/PID.
Currently it's generated based on the HID Descriptor Size. (see [DynamicHID.cpp](../blob/35096a4519466baf79f4e027150044011fd82bd3/src/DynamicHID/DynamicHID.cpp#L73)).

As there's a lot of joystick adapters using arduinos and they all share the same VID/PID, some projects (like the [MiSTer](/MiSTer-devel/Main_MiSTer)) uses the serial to identify different arduino projects.

CDRXavier, 2021-08-22:
> The idea is not to change the "name" that appears on the OS.

You can. If you dig down, waaayy down. Into like USBcore or the like.
It's not as if it will break things -- the device will still work.
> But to change the "serial" that goes along with the VID/PID.

This is actually more difficult.