MHeironimus/ArduinoJoystickLibrary #213: single toggle switch causes multiple inputs

Pac23, 2021-07-24:
## Description of Issue
Single toggle on-off switch causes multiple inputs causing x axis to float.

## Technical Details

- Arduino Board : Pro Micro 
- Host OS : Win 10 same replicated on win 7
- Arduino IDE Version :1.18.15

## Sketch File that Reproduces Issue

The inbuilt example of joystickbutton

## Wiring Details
Pin 10 to positive of toggle switch,pin gnd to negative of toggle switch. 

## Additional context
I tried it on three different computers but the issue persists. arduino wiring is fine because the same wiring works as expected with simhub and mobiflight. 

[image]

CDRXavier, 2021-08-19:
You might need to do pinMode(10, INPUT_PULLUP).

MHeironimus, 2022-03-30:
Are you still seeing this issue?

CDRXavier, 2022-04-01:
I have not have seen this issue and have partially reverse-engineered the code to suit my project.