MHeironimus/ArduinoJoystickLibrary #221: Only 7 axis are recognized by Windows 10. No matter what I do, Acceleratorm Brake and Steering are not recognized.

romandesign, 2021-10-29:
## Description of Issue

Only 7 axis are recognized by Windows 10. No matter what I do, Acceleratorm Brake and Steering are not recognized. 7 axes are working fine: X Y Z Rx Ry Rz Throttle. I can't get any other to appear. Device always appears with a gamepad icon, never as a joystick or anything else. Nor sure if it's relevant.

## Technical Details

- Arduino Board (e.g. Arduino Leonardo): Arduino Leonardo (Pro Micro)
- Host OS (e.g. Windows 10): Windows 10
- Arduino IDE Version (e.g. 1.8.3): 1.8.16

## Sketch File that Reproduces Issue

~~~

#include <Joystick.h> 
#include <AnalogPin.h>  

AnalogPin IN1(A0);
AnalogPin IN2(A1);
AnalogPin IN3(A2);
AnalogPin IN4(A3);
AnalogPin IN5(A10);
AnalogPin IN6(A9);
AnalogPin IN7(A8);
AnalogPin IN8(A7);

Joystick_ Joystick(0x08,JOYSTICK_TYPE_JOYSTICK,
  10, 0,                  // Button Count, Hat Switch Count
  true, true, true,       // X Y Z Axis
  true, true, true,    // Rx Ry Rz
  true, true,           // rudder, throttle
  true, true, true);   // accelerator, brake, steering

void setup() {
   // Initialize Button Pins
  pinMode(15, INPUT_PULLUP);
  pinMode(14, INPUT_PULLUP);
  pinMode(16, INPUT_PULLUP);
  pinMode(7, INPUT_PULLUP);
  pinMode(5, INPUT_PULLUP);
  pinMode(4, INPUT_PULLUP);
  pinMode(3, INPUT_PULLUP);
  pinMode(2, INPUT_PULLUP);
  pinMode(1, INPUT_PULLUP);
  pinMode(0, INPUT_PULLUP);
  //pinMode(A5, INPUT_PULLUP);

  // Initialize Joystick Library
  Joystick.begin(false); //false = dont send automatically. We will sendState() at the end of the loop
  
  Serial.begin(9600);
  //pinMode(LED_BUILTIN, OUTPUT);
  //digitalWrite(LED_BUILTIN, HIGH);
}

// Last state of the pins
int lastButtonState[10] = {0,0,0,0,0,0,0,0,0,0};
  
void loop() {
  
  // Read buttons. 
  Joystick.setButton(0, !digitalRead(0)); //pin 4 LOW means button 0 PRESSED
  Joystick.setButton(1, !digitalRead(1)); //etc.
  Joystick.setButton(2, !digitalRead(2));
  Joystick.setButton(3, !digitalRead(3));
  Joystick.setButton(4, !digitalRead(4));
  Joystick.setButton(5, !digitalRead(5));
  Joystick.setButton(6, !digitalRead(7));
  Joystick.setButton(7, !digitalRead(16));
  Joystick.setButton(8, !digitalRead(14));
  Joystick.setButton(9, !digitalRead(15));
  //Serial.println(!digitalRead(A5));
  
  
  //read analog axes
  IN1.setNoiseThreshold(2);
  int value = IN1.read();
  Joystick.setXAxis(value);
  //Serial.println(value);
  
  IN2.setNoiseThreshold(2);
  value = IN2.read();
  Joystick.setYAxis(value);
  
  IN3.setNoiseThreshold(2);
  value = IN3.read();
  Joystick.setZAxis(value);
  Serial.println(value);

  IN4.setNoiseThreshold(2);
  value = IN4.read();
  Joystick.setRxAxis(value);
  //Serial.println(value);

  IN5.setNoiseThreshold(2);
  value = IN5.read();
  Joystick.setRyAxis(value);

  IN6.setNoiseThreshold(2);
  value = IN6.read();
  Joystick.setRzAxis(value);

  IN7.setNoiseThreshold(2);
  value = IN7.read();
  Joystick.setBrake(value);
  //Serial.println(value);

  IN8.setNoiseThreshold(2);
  value = IN8.read();
  Joystick.setThrottle(value);
  
  Joystick.sendState();
  delay(10);
}  
~~~

## Wiring Details
I can't see more than 7 axis in Windows, so pins are irrelevant

## Additional context
Device always appears with a gamepad icon, never as a joystick or anything else. Nor sure if it's relevant.

positron96, 2021-11-18:
I'm not expert, but have you tried changing joystick type from JOYSTICK_TYPE_JOYSTICK to gamepad or multiaxis controller? Maybe Windows will treat it differently.

romandesign, 2021-11-18:
I tried all available types, including steering wheel etc. Those axes are not recognized. It may well be a Windows limit, but then it should be said so in the docs etc. For now I just specified 2 separate joysticks, and they are recognized as 2 separate devices with a same name by Windows. Just not in a single device.

positron96, 2021-11-19:
Are you checking via Windows control panel? I'm having the impression that that dialog has its own logic of what to show and what not. In particular, it doesn't like Steering/Acceleration/Brake axis (while hidden, they still work fine in a game). Have you tried checking with this website?
https://gamepad-tester.com/

romandesign, 2021-11-19:
I was only checking in Windows control panel. How else would I calibrate those axes, if I can't see them in Windows? It's interesting though that they can still work but be invisible... That could explain it.

eszyman, 2021-12-28:
Hello, I have the same issue.
I can provide this information: Arduino Pro Micro, with the joystick test script running I only get 7 axis.
I confirmed only 7 axis using the "joystick gremlin" program which definitely is capable of seeing more than 7 axis when (vjoy creates an 8 axis controller right from the start) and for confirming more than 32 buttons this works fine as well.
I have been successful in dropping other axis to get rudder, brake, steering, and acceleration to show up, but this is not the desired effect, I would like to have all axis available for my project.

On a possibly related note, I have seen that Windows USB drivers can sometimes override what is detected and that uninstalling and then replugging in USB devices seems to reset things.

This board makes an amazing substitute for the Authentikit joystick project, and I would like to help in any way possible to resolve this issue.  -Thanks!

stevelutz8mesa, 2022-05-26:
Use the example in this library to create multiple joysticks on a single Arduino.  You now have 14 joystick axis available.

PsykomantaBrain, 2023-01-16:
I second Hyratel's solution to simply extend the library definitions to include the two remaining bits, which enables the Dial and Slider axes. I did it here independently and have the same result.

CDRXavier, 2023-02-03:
Disagree. All the axis, the XYZ, RxRyRz, Throttle Brake Steer Rudder Accelerator can be recognized by Windows. See game controller property page for detail.

Whether the SOFTWARE is able to utilize the additional axis(s), is left to individual software. KSP for instance despite being, well, KSP, do not support the additional "simulation controls", even though Windows recognize them.

Zenilogix-Carl, 2026-02-17:
I found a similar problem. It seems that certain combinations of enabled axes work fine, but with some combinations, certain axes will drop out. If I enable Rx,Ry,Rz, and Rudder, then Rudder won't work. If I enable Rx,Ry, and Rudder, it's fine. I can enable all six x,y,z,Rx,Ry,Rz and all work, but if I enable all 11, then Rudder, Steering, Accelerator, Brake drop out. Doesn't look like a consistent pattern other than if any axes are affected, Rudder will be among them. Using joy.cpl, the dropped axes simply don't appear; using DIView, all enabled axes appear, but affected axes don't update, so i looks like joy.cpl uses updates to decide which axes to show. So far as I can tell, the issue is not with the Joystick Library but something in Windows. BTW I am testing on Windows 11 25H2.