MHeironimus/ArduinoJoystickLibrary #222: Will this work on a ESP8266 Nodemcu Board?

simtcrom, 2021-11-01:
Please indicate which board you would like to see supported.
I tried loading the library and trying to flash JoystickTest get below error,

`In file included from C:\Users\sim_t\Documents\Arduino\libraries\ArduinoJoystickLibrary-version-2.0\src/Joystick.h:24:0,
                 from C:\Users\sim_t\Documents\Arduino\libraries\ArduinoJoystickLibrary-version-2.0\examples\JoystickTest\JoystickTest.ino:13:
C:\Users\sim_t\Documents\Arduino\libraries\ArduinoJoystickLibrary-version-2.0\src/DynamicHID/DynamicHID.h:37:28: fatal error: PluggableUSB.h: No such file or directory
   #include "PluggableUSB.h"
                            ^
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
`

positron96, 2021-11-18:
ESP8266 has no hardware USB, so it's not possible to port the library.

MHeironimus, 2022-10-04:
See the following wiki page for the list of supported boards: https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/Supported-Boards

closed 2022-10-04 by MHeironimus