MHeironimus/ArduinoJoystickLibrary #223: Writing code for flight simulator X (DIY Cessna 172 Cockpit)

Cutbangrian, 2021-11-17:
Hi, I want to build flight simulator cockpit for cessna 172.
turned out Saitek yoke is currently at $200 +

So I ended up here.
I'm using arduino Leonardo and succesfully tested joysticktest.ino
all button is verified and tested when I grounded A0.

then I saw code in this documentation called flightcontrolledtest.ino
but I dont know how to use it for my DIY yoke for simulator.

I have read the code more than 50 times but dont understand how to wire it up my potensiometer to arduino pin.
because code in flightcontrolledtest.ino is so complicated.
but I really like flightcontrolledtest.ino code because it contains X Y axis, rudder, throttle and 32 button dan I can use for landing light, etc in cessna 172.

Im really newbie at coding, I can understand just a little bit.
I already searched at youtube and couldnt find anyone that show how to do it.
even if there is exist, they only provide XY axis and throttle.

but in this case, I need other 32 button like flightcontrolledtest.ino code.

how I can overcome my problem?

schnoog, 2021-11-23:
Wiring up
https://www.arduino.cc/en/Tutorial/BuiltInExamples/AnalogReadSerial

initialize the joystick  (example for X and Y axis  and throttle of an joystick)
`Joystick_ Joystick(0x03,
  //  joystickType          buttonCount   hatSwitchCount  
  JOYSTICK_TYPE_JOYSTICK,     32,            0,
  //includeXAxis includeYAxis includeZAxis includeRxAxis includeRyAxis includeRzAxis
       true,         true,       false,       false,        false,        false,
  //includeRudder  includeThrottle includeAccelerator includeBrake includeSteering
       false,           true,            false,            false,       false);
;`

define the range in "void setup"
`Joystick.setYAxisRange(0, 1023);
Joystick.setThrottleRange(0,1023);
`

in "void loop" you have to read the values and assign them to the axis, and "submit" the control input
`
        pinMode(A0, INPUT);
        pinMode(A1, INPUT);
        pinMode(A2,INPUT);
        int AX =  analogRead(0);
        int AY = analogRead(1);
        int THROT = analogRead(2);
        Joystick.setXAxis(AX);
        Joystick.setYAxis(AY); 
        Joystick.setThrottle(THROT);
        Joystick.sendState();
`

Cutbangrian, 2021-11-23:
OMG finally my yoke is released and completed.

A huge thanks to you my friend.
I can't expressed it.

Once again, thank youu much !!!

[image]

schnoog, 2021-11-24:
You're welcome. 
A little change to the code
the 3 lines
`pinMode(A0, INPUT); pinMode(A1, INPUT); pinMode(A2,INPUT);`
should be in the "void setup" and not "void loop".

The current should not cause any issues, so more a cosmetic aspect.

closed 2022-03-30 by MHeironimus