MHeironimus/ArduinoJoystickLibrary #225: Problem with Flight Simulator 2020 or Xplane 11

Pietrogramma, 2021-12-01:
Hi everybody, I would use this sketch for my Home Cockpit that I use with MFS2020 or Xplane11, but I have a problem if I use a switch button.
if I connect a switch to the arduino micro and assign a function of the flight simulator to this switch (for example taxi light) I see that if I lower the lever of the real switch even in the simulator the lever of the taxi light is lowered, when I raise up the lever of the switch real, the simulated one does not react, it stays down. 
If I lower the real one again, the simulated one finally gets up. 
In other words, the virtual lever reacts to every two commands of the real one.
There is a way for solve thisi bug?
Thanks and ciao

SimonKendrick99, 2022-01-20:
Sounds like you have the input set to toggle. As if you just had a key to turn on or turn off a function. Not at computer now but you can change the input type when mapping.

Pietrogramma, 2022-03-31:
> Are you still seeing this issue?

Yes I don't know how I  must change the Input how suggest SimonKendrick99, Can you help me, please?
Thanks and ciao