MHeironimus/ArduinoJoystickLibrary #227: Can we have this to support Arduino NANO?

Harrastaja, 2021-12-13:
I will get now info as following:
In file included from C:\Users\JouniHeinonen\Downloads\RCUSB4\RCUSB4\RCUSB4.ino:9:0:
C:\Users\JouniHeinonen\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/Joystick.h:32:2: error: #error The Joystick library can only be used with a USB MCU (e.g. Arduino Leonardo, Arduino Micro, etc.).
 #error The Joystick library can only be used with a USB MCU (e.g. Arduino Leonardo, Arduino Micro, etc.).
  ^~~~~
exit status 1
Error compiling for board Arduino Nano

Is this possible to add support for Nano? or is this hardware base issue?

Harrastaja, 2021-12-16:
I have notice that this issue comes from hardware. so this is close.

closed 2021-12-16 by Harrastaja

Raphfriend, 2021-12-16:
This project use as base the ATmega32u4, nano is 328p, so probably no.