MHeironimus/ArduinoJoystickLibrary #228: Howto use a rotary ?

xXThanatosXx997, 2021-12-25:
Hello Guys i am making a controller and i want to use this library. I have attached buttons succesfully but i also want to add a rotary. Any idea how to do it ? is it supported ?

cagri-yuksel, 2021-12-27:
You have to use rotary encoder library thats makes your job more easier.

xXThanatosXx997, 2021-12-27:
Could you sent me a link to the library ? also is there any example with interupts anywhere ?

cagri-yuksel, 2021-12-27:
There so much rotary encoder libary. Sorry i am really far away my house right now. I can help you 2 week later if you need still help. 
You can use this https://github.com/mathertel/RotaryEncoder
Or
https://github.com/Erriez/ErriezRotaryEncoderFullStep
https://github.com/Erriez/ErriezRotaryEncoderHalfStep
You can use this methods i guess, 
Check rotary encoder position  and make if statement. 
Like if rotary encoder is cw press button 7 and release or ccw press 8 simple but you have to figure out how its work. 
Joystick.pressButton(uint8_t button)
Add delay or sens from library. My advice add delay 20 between 100ms
Joystick.releaseButton(uint8_t button)
Add delay short like 50.
Otherwise you can use stm32f103c8t6 based FreeJoy really easy to use and large community they will answer your question at dc.
I make my hotas with freejoy.

laserbeak, 2022-01-18:
> Otherwise you can use stm32f103c8t6 based FreeJoy really easy to use and large community they will answer your question at dc. I make my hotas with freejoy.

Arduino isn't compatible with freejoy, right?

cagri-yuksel, 2022-01-18:
No but you can use MMjoy.

MHeironimus, 2022-03-30:
This library does not focus on how to read values from input devices. It only focuses on making the Arduino appear to the host device as a game controller / joystick. Hopefully the comments above will help.

closed 2022-03-30 by MHeironimus