MHeironimus/ArduinoJoystickLibrary #229: Using Hall effect Sensors.

Ticskter, 2022-01-02:
Hello, i am new to GitHub and this Arduino programming stuff. i am a TOTAL noob at this and need some assistance please.

i purchased a kit to build my own American truck Simulator Shifter that has Hall Effect Sensors that register Range Splitter and Gear Splitter settings as buttons. The company (single person) i purchased the kit from does not program the board and will not "release" his code as it is his full time job making shifters for ATS simulator, which i understand. He also makes fully assembled kits that are plug and play but there is nearly a year back log of orders. i originally ordered an assembled kit in March of 2021 and was supposed to receive it in  August 2021. Unfortunately he was nearly 300 orders behind so he began offering DIY kits.

i have an Arduino Micro Pro that i have used this library to install so that windows can see it as a game controller and it does so perfectly, however, when i place a small magnet that is provided in the kit next to the hall effect sensors there are no buttons showing being pushed in the windows controller settings window.

i followed a video of how the guy has the Board and Hall Sensors are wired but it wasnt a very high quality video and it was sped up as a time lapse so i played it as slow as i could and paused to see where the wires were in the video.

i have attached an image of a wiring diagram for reference as i read something about if pin 9 is grounded etc. and like i said earlier im a total NOOB at all this stuff.
[image]

I used the "joystick button" sketch because all that is needed from this board are 2 buttons and that is it so i thought that would be the best bet for my situation. I dont know if that code is even written to see Hall Effect Sensors or not so i dont know if it was the proper choice. if it will read the Sensors then i will gladly rewire the board so that it works!!!! If there has to be a custom code in order to read the signal from the Sensors would someone kindly be of help and write it so that it works or add the option to the current sketch library? if not, then where would be the best place to go to find the proper code?

cagri-yuksel, 2022-01-12:
I don't understand your question. You try to ask how to wire hall effect sensor to arduino and read data with joystick libary? Then everyting same as potentiometer. Search outputs your hall effect sensors and wire to arduino. Like this photo. 
[image]

MHeironimus, 2022-03-30:
This library does not focus on how to read values from input devices. It only focuses on making the Arduino appear to the host device as a game controller / joystick.

You need to write (or find) the code that reads the Hall Effect Sensors. Once you have read those values, you can use those values when calling this library.