MHeironimus/ArduinoJoystickLibrary #233: Request buttons

JerryDaga, 2022-01-29:
Good day sir.
I have requested a new code for all types of Arduino.
Can you make it as "Piezo Sensor to Button Switch or Mapping"?

I want to make a D.I.Y. Table Top E-Drum like this (Attached Photo)

[Scan - 2022-01-29 19_02_19.pdf](https://github.com/MHeironimus/ArduinoJoystickLibrary/files/7963544/Scan.-.2022-01-29.19_02_19.pdf)

fbrinker, 2022-03-15:
I am not sure if I understand your request correctly, but you should be able to map that signal to multiple buttons.
Just connect the sensor to an analog input and use the arduino function https://www.arduino.cc/en/Reference/AnalogRead to get the sensors value.

Then you can simulate a button press via Joystick.pressButton(x) if the value is < 100. Another if the value is < 200, < 300, < 400... as an example. And depending on the sensors intensity another button is pressed. Or you can map all inputs above 0 to a single button.

I've never worked with a piezo sensor, so I can't recommend any code basics. You may find some help over in the Arduino forums :)

MHeironimus, 2022-03-30:
This library does not focus on how to read values from input devices. It only focuses on making the Arduino appear to the host device as a game controller / joystick.

closed 2022-03-30 by MHeironimus