MHeironimus/ArduinoJoystickLibrary #235: Serial Port and Joystick library on Leonardo

AdriGuillaume, 2022-02-12:
Hello,

I would like to use temporaly Serial port for monitoring my Leonardo before initiliazing the joystick class in Setup().
but it seems that Serial.begin() disable definitly joystick HID for the rest of my program even if i use Serial.end() at the end of my instruction.

someone can help me ?

Thanks a lot ! 
Adrien

fbrinker, 2022-03-14:
It seems to work for me. I use the Serial monitoring for debugging purposes.

My setup looks like the following:
```
void setup()
{
   Serial.begin(115200);
//...
  Joystick.begin(false);
//...
```

And the loop:
```
void loop()
{
  Serial.println("Hello");
//...
  Joystick.sendState();
```

Note that I don't use the auto-sending of Joystick states. But I guess that shouldn't be the problem. But you may want to give it a try.

GO63-samara, 2022-03-24:
You can also do this:

You need to use `Serial1` instead of `Serial`.
Then debug will go through the TX and RX pins and through the USB-UART module.
HID will work independently.

You can put in the first line:
`#define Serial Serial1
`

MHeironimus, 2022-03-30:
Question appears to be answered above.

closed 2022-03-30 by MHeironimus