MHeironimus/ArduinoJoystickLibrary #236: Unable to deal with more than 2 controllers in a single sketch

guillaumeparent, 2022-03-21:
## Description of Issue
As stated in the title, I am unable to create a sketch able to set more than 2 controllers.
The discussion started in [Mathiew's blog](http://mheironimus.blogspot.com/2016/11/arduino-joystick-library-version-20.html?showComment=1647775571467#c6579748545785113185).

To summarize, I manage to create sketches dealing with 2 controllers, but as soon as I try to deal with more than 2 controllers only the 2 first ones are properly defined.

I attach in the following a minimal reproducible example attempting to define 4 controllers with 1 button each.
As one can see on the attache picture, only the 2 first (0x10 and 0x11) controllers are properly detected. The 2 last ones (0x12 and 0x14) are not. The axis are recognized as buttons only.

[image]

## Technical Details

- Arduino Board: Arcuino Micro and Arduino Pro Micro: 
- Host OS: MacOS (Catalina)
- Arduino IDE Version: 1.8.19
- Joystick library version: 2.0.7

## Sketch File that Reproduces Issue

~~~
#include <Joystick.h>

#define NbrGamePad 4

Joystick_ Joystick[NbrGamePad] = {
Joystick_(0x10, JOYSTICK_TYPE_GAMEPAD, 1, 0, true, true, false, false, false, false, false, false, false, false, false),
Joystick_(0x11, JOYSTICK_TYPE_GAMEPAD, 1, 0, true, true, false, false, false, false, false, false, false, false, false),
Joystick_(0x12, JOYSTICK_TYPE_GAMEPAD, 1, 0, true, true, false, false, false, false, false, false, false, false, false),
Joystick_(0x13, JOYSTICK_TYPE_GAMEPAD, 1, 0, true, true, false, false, false, false, false, false, false, false, false)
};

void setup() {
// put your setup code here, to run once:

}

void loop() {
// put your main code here, to run repeatedly:

}
~~~

## Wiring Details
Any pin wiring details that may be relevant.

## Additional context
Add any other context about the problem here.

MHeironimus, 2022-03-21:
The sketch file above appears to work correctly on Windows 10. It may be a MacOS issue.

For example:

[image]

First Controller:

[image]

Fourth Controller:

[image]

The second and third look the same.

I know Linux machines need an adjustment to work correctly if more than one controller is defined (see https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/FAQ#linux-support for more details). Maybe there is something equivalent for the Mac.

guillaumeparent, 2022-03-21:
Ok, I found the problem. It was indeed the issue mentioned [here](http://mheironimus.blogspot.com/2015/09/linux-support-for-arduino-leonardo.html).
I tested the method on a Raspberry Pi (on which I experienced the same 2 controller limitation) and it worked. Now I can see 4 controllers.

Thank you for the help !

closed 2022-03-21 by guillaumeparent