MHeironimus/ArduinoJoystickLibrary #240: Does not compile for Teensy4.1

bwanaaa, 2022-05-02:
## Description of Issue
Including the Heironimus ArduinoJoystick library in an otherwise empty sketch causes a compilation error
## Technical Details

- Teensy 4.1
- Windows 10):
- Arduino IDE Version (1.8.19):

## Sketch File that Reproduces Issue

~~~
#include <Joystick.h>
void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
~~~

Error that arduino IDE reports:

```
In file included from C:\Users\Stefan\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/Joystick.h:24:0,
                 from C:\Users\Stefan\Documents\Arduino\teensy\joystick\joystick.ino:1:
C:\Users\Stefan\Documents\Arduino\libraries\ArduinoJoystickLibrary-master\src/DynamicHID/DynamicHID.h:37:28: fatal error: PluggableUSB.h: No such file or directory
compilation terminated.
Error compiling for board Teensy 4.1.
```

MHeironimus, 2022-05-02:
Teensy is not supported. See https://github.com/MHeironimus/ArduinoJoystickLibrary/wiki/Supported-Boards for more details.

bwanaaa, 2022-05-04:
thank you. it seems i am still learning to read.

closed 2022-05-04 by bwanaaa