MHeironimus/ArduinoJoystickLibrary #254: Problem with Joystick.sendState() in setup()

mistepien, 2022-10-31:
This is code for simple joystick for vice emulator. In setup() I wanted to set axis X and Y in center. It works only with huge delay().

`Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_JOYSTICK,
                   3, 0,            // Button Count, Hat Switch Count
                   true, true, false,     // X and Y, but no Z Axis
                   false, false, false,   // No Rx, Ry, or Rz
                   false, false,          // No rudder or throttle
                   false, false, false);  // No accelerator, brake, or steering

void setup() {
  Joystick.begin(false);
  Joystick.setXAxisRange(-1, 1);
  Joystick.setYAxisRange(-1, 1);
  delay(2000); //very ugly and dirty hack
  Joystick.setXAxis(0);
  Joystick.setYAxis(0); 
  Joystick.sendState();`

If I remove delay(2000) than joystick is not set to center.

GonzoG, 2023-01-10:
You should set axes range before calling Joystick.begin();

mistepien, 2023-01-10:
It does not work for me. Without "delay(2000)' joystick is set to (-32767,-32767), what is UP-LEFT corner.

`#include <Joystick.h>

Joystick_ Joystick(JOYSTICK_DEFAULT_REPORT_ID, JOYSTICK_TYPE_JOYSTICK,
3, 0, // Button Count, Hat Switch Count
true, true, false, // X and Y, but no Z Axis
false, false, false, // No Rx, Ry, or Rz
false, false, // No rudder or throttle
false, false, false); // No accelerator, brake, or steering

void setup() {
Joystick.setXAxisRange(-1, 1);
Joystick.setYAxisRange(-1, 1);
Joystick.begin(false);
//delay(2000); //very ugly and dirty hack
Joystick.setXAxis(0);
Joystick.setYAxis(0);
Joystick.sendState();
}

void loop() {
  // put your main code here, to run repeatedly:

}`